aritchie/bluetoothle #3: How to know when read operation is complete?

rlingineni, 2016-11-22:
Hi, 

I am trying to read a string value from a characteristic. How can I do that?

            StringBuilder read = new StringBuilder();

		var reading = characteristic.Read().Subscribe(bytes => { 
							var a = Encoding.UTF8.GetChars(bytes); 
							read.Append(a); 
						});`

I have tried subscribing and then writing to a char array, and am building a StringBuilder object. I don't think I am doing this the right way. What is the best way to read values from a characteristic?

aritchie, 2016-11-22:
You are doing it right if your end result is equal to the MTU size (20 bytes by default).  There isn't really the concept of flow control for messages stretched across multiple packets.  You have to build this functionality yourself.  

The value in read is considered a completed operation so you have to loop if you are expecting more

rlingineni, 2016-11-22:
Ah ok, got it. Thank you!

rlingineni, 2016-11-22:
`var x = characteristic.Read(); `
Say I just want to read a value, is it possible I can do so without subscribing and just read everything at once? 

I see how everything is an `IObservable<byte[]>`what exactly does that mean? How do I know when the reading is complete?

rlingineni, 2016-11-22:
Actually, I think I kind of got it, but not completely.

 I was just going to through the docs, and I saw that there was an `await` version for read. However, the `read` method doesn't seem to be an async call. If I can await the read operation, then I should be good to go.

aritchie, 2016-11-22:
All observables are awaitable via reactive extensions but only when you expect a single result.  In the case of read/write, it is perfectly fine to await them since they are completed singular operations.

It should be noted that it is also a good practice to stick a timeout on it

aritchie, 2016-11-25:
Use IGattCharacteristic.ReadUntil in 1.0.8, it expects some sort of ending packet like 0x0 so it knows where to complete

closed 2016-11-25 by aritchie