aritchie/bluetoothle #4: When is version 2 or a beta -- need UWP Windows 10 Phone

hubert-associates, 2017-01-04:
When is version 2 -- need UWP Windows 10 Phone. Yes the phone lives :-)

closed 2017-01-07 by aritchie

hubert-associates, 2017-01-11:
Thanks. Not expecting a real timeline, just wondering. I've got the UWP Windows 10 Phone sample running -- or most of it -- now. I noticed that if one starts the samples on a machine without a bluetooth adapter that it crashes with null pointer exceptions in a few places. I'll see what I can do, but I'm still getting familiar with this. No timelines :-)

SoyPan, 2017-01-13:
Apparently they just updated the SDK (build 15003 and higher) with new APIs that should help you solve most of the problems and ugly workarounds with UWP and BLE. 

http://aka.ms/bluetoothgatt