aritchie/bluetoothle #5: No services for Android 7.0

StefaanAvonds, 2017-01-19:
I'm not sure if it is related to the Android version, but my Google Nexus 6 Android version 7.0 is not able to discover all the services of a Bluetooth LE-device.

For example: when connecting to a PunchThrough Bean-device I can only discover 1 service instead of the usual 10 (service ID: F000FFC0-0451-4000-B000-000000000000).

aritchie, 2017-01-19:
Please share your discovery code.  Also, do other BLE scanner apps pick up the additional services?

StefaanAvonds, 2017-01-19:
The Monkey Robotics plugin gives the same result: only 1 service found. Another device with Android 6.0.1 does find the other services.

The code:
```
Device.Connect().Subscribe();
Device.WhenServiceDiscovered().Subscribe(service =>
            {
                var desc = service.Description;
                service.WhenCharacteristicDiscovered().Subscribe(characteristic =>
                {
                    var bytes = characteristic.Value;
                });
            });
```

aritchie, 2017-01-19:
There is no problems I can see with your code.  As requested, please try another actual scanner app and see if it produces the services you think should be present.

StefaanAvonds, 2017-01-19:
I've tried the app "[Bluetooth LE Scanner](https://play.google.com/store/apps/details?id=uk.co.alt236.btlescan)". The Google Nexus 6-device can only find 1 service, while the other Android-device can find everything.

Are there some settings I need to activate on this device in order to find the other services?

aritchie, 2017-01-19:
This kind of goes outside of a library problem then.  If two libraries and a scanner app are only picking up one service on the particular phone, it sounds like a problem with the phone.  

As for settings, not that I know of.

closed 2017-01-28 by aritchie