aritchie/bluetoothle #6: Force notification subscription

SoyPan, 2017-01-25:
Hello, I was wondering if there is a way to force a notification subscription even if I am getting CanNotify as false, while using the NuGet.

As a background, I am using Microchip's Low-energy Data Protocol (MLDP) which is a private service/characteristic, and I am getting CanNotify false even though I know for sure it can notify because it works with an iOS app we previously designed. Unfortunately, I have not found a way to fix the issue with the flag on the BLE adapter end (I think it is some issue with how Microchip's private service is configured), and I'd rather not spend time making my own service at the moment.

I can get everything to work in both Android and iOS with this (fantastic btw) library by going to Extensions.cs and forcing CanNotify to return "true" every time by cloning and compiling along my app, but I'd rather use the NuGet if possible.

Thank you very much in advance!

aritchie, 2017-01-25:
What are the actual bluetooth characteristic properties on the characteristic you are trying to "force"?  Does it contain "Indicate" as a property.  Just print out the following: characteristic.Properties.ToString()

SoyPan, 2017-01-25:
It outputs "Write, Indicate". I noticed you added "indicate" as an option for the CanNotify() flag to the latest 2.0 commit, so that should do it, thanks!. Are you planning on pushing that to the prerelease NuGet soon? If not, I'll add it as a submodule, so no biggie. 

Thank you very much for your help!

P.S. Weird thing is we ordered many of these devices (RN-4020), and some do output "Read, WriteNoResponse, Write, Notify, Indicate". I'll have to take a deeper look at that.

aritchie, 2017-01-25:
I am planning to push to nuget within the next few days.

SoyPan, 2017-01-25:
Alright, sounds perfect. Thanks!

aritchie, 2017-01-28:
fixed in 2.0

closed 2017-01-28 by aritchie