aritchie/bluetoothle #7: Cannot see IsMtuRequestAvailable and RequestMtu.

oOGabsOo, 2017-02-06:
Hello,

I just installed the package with nuget. But when I receive a ScanResult and try to access the device in order to try a RequestMtu, the methods are not available in my form project. 

` 
BleAdapter.Current.Scan().Subscribe(scanResult => {

                scanResult.Device.Connect();
                scanResult.Device.IsMtuRequestAvailable //Doest not contain a definition....blabla
});
`

What am i missing here? I saw in the sample in the repository that you are actually accessing it without any trouble in "DeviceViewModel" : 

` 
 this.RequestMtu = ReactiveCommand.CreateFromTask(
                async x =>
                {
                    if (!this.device.IsMtuRequestAvailable)
                    {
                        this.Dialogs.Alert("MTU Request not supported on this platform");
                    }
` 
Thank you in advance for your help.

Cheers,

aritchie, 2017-02-06:
what nuget package version are you using?

oOGabsOo, 2017-02-06:
Ok solved ;) It was a missing update of the package ;)

closed 2017-02-06 by oOGabsOo