aritchie/bluetoothle #11: Android Connection Cleanup

aritchie, 2017-02-07:
@ajchacko I'll fix this for beta4 tonight.  I've posted this here for you to track.  

Disconnect is not enough.  Close cleans up all of the left over junk.  It wasn't a problem in later android releases.

ajchacko, 2017-02-07:
Awesome thank you!!

aritchie, 2017-02-08:
@ajchacko I took your advice about the disconnect and used that when cancelling connections.  When dispose is called, I clean up the whole thing.

I haven't had a chance to test this yet (the change is minor).  I wanted to make sure I got this in for you as soon as possible though.

ajchacko, 2017-02-15:
Thanks Allan, that seems to have fixed the problem. One thing I noticed is that the Dispose method for Device & AbstractDevice is not in its interface IDevice. So I don't think there is currently a way to call the Dispose method to clean up the connection. I can create a pull request if its easier. Thx.

aritchie, 2017-02-16:
I don't really want the end user disposing of a device.  In any case, it isn't leaking memory at the moment and it is working for you, so let's just let this one be for now.

closed 2017-03-04 by aritchie