aritchie/bluetoothle #8: RSSI instead of WhenHeartRateBpm

Aphire, 2017-02-07:
First off, great plugin, I really appreciate the work that has gone into it.

I've been toying around with the heart rate monitor plugin and i've found that my device returns true for 
device.HasHeartSensor();

However nothing pings for:
WhenHeartRateBpm()

Instead I am able to pull what I assume to be the heart rate from:
WhenRssiUpdated()

This is great for the single device, however the app is to be used with a variety of devices, will they all behave in this way? Or will the HeartRateBPM ping on some, RSSI on others? What would you suggest?

Also may be worth adding this information to your readme as I wasted a fair bit of time scratching my head over this.
Thanks!

aritchie, 2017-02-07:
RSSI & Heartrate have absolutely nothing to do with each other.  RSSI is received signal strength.  Heart rate depends on what bluetooth sensor your using to monitor the human heart, so I'm not sure what you are expecting me to document?

Aphire, 2017-02-07:
Ah sorry, I was receiving data from the RSSI that was in the range of resting heart rate (upper 80's), leading me to believe that's what I was receiving which is why I said "What I assume to be heart rate".

Then in which case, I'm not getting any heart rate data from WhenHeartRateBpm(), however when I use the same heart rate monitor with another app, it appears to work as expected.

Perhaps StackOverflow would be a better place to pose this question, so I can dump some code.
Feel free to close this issue, but if you find the time, please also take a look over my StackOverflow question.

aritchie, 2017-02-07:
Keep in mind, the heartrate functions are not a one size fits all.  You should figure out what services/characteristics you are hooking to get proper functionality on the BLE device you are using.  I won't be able to help you at all (neither will SO) because we don't know the device specifics.  You'd have to review what docs they have.

Also, not sure how I'm supposed to answer a SO question when I have no where it is landing.

closed 2017-02-07 by aritchie