aritchie/bluetoothle #12: Can't find any smpartphones.

Werner91, 2017-02-15:
Hi,

I'm using your tool for a couple hours now and I could not fix a problem.
In my Xamarin App I'm just scanning for alle bluetooth devices around me.
It shows bluetooth devices like Smartwatches and Headphones but I can't find my other Smartphones.
If I'm checking alle bluetooth devices in the bluetooth settings there are all other smartphones and devices.

Is this plugin just for bluetooth equipment but not for Smartphones?

aritchie, 2017-02-15:
It is called bluetooth LE for a reason :)  

Do you have a BLE server running on your smartphone?  This also only sees LE/Dual based devices, not classic bluetooth based devices.

Werner91, 2017-02-15:
Thanks for your fast response. :)
Well I tought "Bluetooth = Blutooth"  but I see thats to easy with a NuGet ;) Would this work if I use my smartphone to deal with a NFC scanner pen? 

Do you know any usefull tutorials für Xamarin(.Forms) for classic bluetooth? Classic bluetooth should also work with a NFC scanner pen?

aritchie, 2017-02-15:
NFC != Bluetooth (classic or le). NFC = near field communication.  

As for an article, no idea.  Classic bluetooth truly only works on android & UWP.  You will pretty much be out of luck for iOS.  NFC is 100% out of iOS.

Seeing as these questions are way outside of this library, I'm going to close this out.  Good luck on your solution.

closed 2017-02-15 by aritchie