aritchie/bluetoothle #13: No longer scans on iOS

betty-crokker, 2017-02-15:
With the latest version available on nuget (1.3.1), this library no longer finds anything when scanning for BLE devices on an iOS device (iPad).  Still works fine for Android.

aritchie, 2017-02-16:
Does the sample work for you? Can you provide a reproducible sample?

betty-crokker, 2017-02-16:
public partial class SelectBluetoothDevicePage : ContentPage
    {
        System.IDisposable m_scanResults = null;

        protected override void OnAppearing ()
        {
            m_scanResults = BleAdapter.Current.Scan ().Subscribe (scanResult => Debug.WriteLine ("BLE " + scanResult.Device + " RSSI=" + scanResult.Rssi));
        }

    }

betty-crokker, 2017-02-16:
I'm 99% certain this iPad can talk BLE, it's a model MBL12LL/A which means it's an iPad Air 2, and according to [this page](http://www.iphonefaq.org/archives/973945) it can do BLE.  I'm charging an iPad model [A1454](http://www.everymac.com/systems/apple/ipad/specs/apple-ipad-mini-original-a1454-4g-lte-att-specs.html) now to test my program there, it does Bluetooth 4.0.

aritchie, 2017-02-16:
I'm glad you've solved your issue.  In any case, I'll add an exception to the scan call if the adapter isn't ready

closed 2017-02-17 by aritchie