aritchie/bluetoothle #14: ScanListen vs. IsLowPoweredScan=True

v9rt3x, 2017-02-16:
Hey Allan,

thanks for your great work!
I have some questions:
1. What's the difference between:
BleAdapter.Current.**ScanListen()**.Subscribe(scanResult => { Debug.WriteLine(scanResult.Device.ToString()); });

**and**

BleAdapter.Current.Scan(new ScanConfig
                {
                    **IsLowPoweredScan = true**,
		    ServiceUuid = new Guid("00000000-0000-0000-0000-xxxxxxxxxxxx")
                })
                .Subscribe(x => {});

2. With ScanListen() i don't get any results.. why?
3. With the second procedure i get results if i dont add the constraint to a specific Uuid(ServiceUuid = new Guid("UuidString")). I also tried it with Guid.Parse(""). Without any results. Which Uuid do u provide to listen to a specific device? The Device.Uuid or the one which is listed under AdvertisementData.ServiceUuids ? I tried both without any results.

Thank you for your thoughts!

aritchie, 2017-02-16:
1) ScanListen isn't in the docs, but it is documented on the interface: "Allows you to listen in on current scans in progress (usualful for background tasks like logging and decoupled components)"  

Just as an FYI, it will be going away in v2 which is due to be released in the next week or so.  Just stick to Scan().

2) You likely don't have an existing Scan() call being observed.  Scan actually starts the scan process and then hooks to ScanListen() internally.  

3) The service UUID filters on the advertisement data service UUIDs.  If you want to look for a specific device, use something like BleAdapter.Current.Scan().Where(x => x.Device.Uuid.Equals(...));

If you are having trouble with the service UUID filter itself, please file a secondary defect this thread isn't hijacked please.  Please also include the platform that is giving you the problem as well.

4) IsLowPowered is used to not hit the CPU as hard during a scan.  It only has an effect on Android.

v9rt3x, 2017-02-16:
ok, thx!
I intended to use the "IsLowPowered-scan" for devices which don't provide coupling to save battery capacity. If this is the correct way to do that the issue can be closed.

I will open a new thread for my service UUID filter problem.

aritchie, 2017-02-16:
It is, but you really don't need it.  My suggestion is not to use low powered scanning unless you are doing the scanning in the background (which is extremely difficult in the background on iOS) or continuously on the foreground.

closed 2017-02-16 by aritchie