aritchie/bluetoothle #15: ServiceUuid-Filter

v9rt3x, 2017-02-16:
Hey Allan,

Plattform: Nexus 5
Android Version: 6.0.1

**What I did**:
A scan collecting all offered service uuids.

foreach(Guid objGuid in x.AdvertisementData.ServiceUuids) {
Debug.WriteLine(objGuid.ToString());
}

Mostly each device offers only one service.
With this service-uuid i tried to subscribe this specific service to listen continuously to this service. 
ScanConfig{
ServiceUuid = new Guid/Guid.Parse("xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx")
}

Maybe i misunderstand the intention of this filter and just should use the where constraint but what is then this filter for?

Thx again for your answer!

aritchie, 2017-02-17:
You have the right idea.  Are you able to share one of the missed UUIDs?  I just need to know the size of the UUID that may be missed.

v9rt3x, 2017-02-17:
service-uuid example: 6ebbdb13-605a-afd3-f1a7-4dffff3e315c

**Off-Topic question**(hope it's ok, because it's no bug): If i do a scan i get continuously advertisement data (after some time also from one device twice). Is there an opportunity to scan in a way to just get each device once? Or do I have to manage that manually through checking my "device holder" if a specific uuid already exists?

thx in advance!

aritchie, 2017-02-17:
Thanks.  I'll take a look

As for the device holder, it is something you would need to do.  With observables though, you could do something like this (though you won't get any more scan results for it):

var scan = BleAdapter.Current.Scan().Distinct(x => x.Device.Uuid).Select(x=>x.Device).Subscribe(device => ...);

v9rt3x, 2017-02-17:
thanks a lot Allan, helped me a lot!

aritchie, 2017-03-04:
V2.0 that I released last night should have some fixes around the service uuid scan.  I never had the chance to verify the fix though.  There is also a new method method BleAdapter.Current.ScanForUniqueDevices that accomplishes your other question

v9rt3x, 2017-03-06:
Thank you!
I don't get it to work anyway. But I can't also exclude that it is an issue from the sensor itself. I'll try it with another one and let you know.

I also have a question which doesn't affect this library directly though is hard-wired.
Is it possible to send you a private message/email?

kind regards

aritchie, 2017-03-06:
The service UUID scan filter was broken.  It was not your sensor.  

As far as contacting me directly, uou can reach me on the xamarin slack channel at @aritchie and I respond as I get a chance

closed 2017-03-11 by aritchie