aritchie/bluetoothle #16: iOS scan not finding BLE-device

StefaanAvonds, 2017-02-17:
In the Info.plist I've added:

```
<key>UIBackgroundModes</key>
	<array>
		<string>bluetooth-central</string>
	</array>
  <key>CFBundleName</key>
  <string>Bluetooth</string>
  <key>NSBluetoothPeripheralUsageDescription</key>
  <string>Bluetooth is needed</string>
```

But the iOS-device is not scanning:

```
BleAdapter.Current.Scan().Subscribe(result =>
            {
                // Not working...
            });

BleAdapter.Current.Scan(new ScanConfig { ServiceUuid = idService, IsLowPoweredScan = true }).Subscribe(scanResult =>
            {
                // Also not working...
            });
```

Is there another setting that needs to be activated?

aritchie, 2017-02-17:
Keep a reference to the scan.  var scan = BleAdapter.Current.Scan().Subscribe();

If that still doesn't work for you, please post a reproducible sample and I'll take a look.

StefaanAvonds, 2017-02-17:
I tried that, but no success. Please take a look at the solution in the current ZIP-file and let me know what I'm doing wrong.

[Hackaton.zip](https://github.com/aritchie/bluetoothle/files/783017/Hackaton.zip)

aritchie, 2017-02-17:
I believe the service UUID filter isn't working as intended.  This is actually a duplicate #15 

However, I don't see a problem with the normal scan.  Do the samples work for you?

StefaanAvonds, 2017-02-17:
It seems like the adapter doesn't scan when given the Advertisement ID in the ScanConfig. I'm trying to only scan for devices with a certain Advertisement ID.

I solved it by checking manually per device.

`if (scanResult.AdvertisementData == null || scanResult.AdvertisementData.ServiceUuids == null || scanResult.AdvertisementData.ServiceUuids.Count() == 0 || scanResult.AdvertisementData.ServiceUuids[0] != Bean.BeanServiceAdvertisingUuid) return;`

aritchie, 2017-02-17:
Thanks for the follow up.  I'll take a look into this.  

I'm going to close this issue.  Please #15 for updates on your issue

closed 2017-02-17 by aritchie