aritchie/bluetoothle #17: No services or characteristics found on Android

StefaanAvonds, 2017-02-20:
Android device: Google Nexus 6
OS: Android 7.0

Once connected to a specific Bluetooth LE-device ([Bean from Punchthrough](https://punchthrough.com/bean)) the adapter never discovers any service or characteristic.

**Code to scan**
```

Adapter.Current.Scan().Subscribe(result =>
{
    // Add found device to ListView
});

```
**Code to connect**
```

device.Connect().Subscribe(result =>
{
    // Specific code
});

device.WhenStatusChanged().Subscribe(status =>
{
    if (status == ConnectionStatus.Connected)
    {
        // Navigate to new page
    }
});

```
**Code to scan for service and characteristic AFTER navigation to page**
```

var serviceDiscover = device.WhenServiceDiscovered().Subscribe(service =>
{
    // Never enters this code block
});

var characteristicDiscover = device.WhenAnyCharacteristicDiscovered().Subscribe(characteristic =>
{
    // Never enters this code block
});

```

What am I doing wrong?

aritchie, 2017-02-20:
Do the samples work for this ble device info n your android device?

StefaanAvonds, 2017-02-20:
In the samples I can connect with the BLE device, but no characteristics.

aritchie, 2017-02-20:
Please try the 2.0 beta in Nuget.  The device I use for testing is working fine, so I'm not sure what the issue is at the moment.  I don't have the time right now to be debugging two versions so you could be waiting a while for a fix

StefaanAvonds, 2017-02-20:
In the 2.0 beta version it's still not working... Same problem: no problem when connecting but no service or characteristic is discovered.

aritchie, 2017-02-20:
This is a duplicate of #5 which you already filed.  You said you were having difficulties with other scanners too.  I have been unable to reproduce the problem on my end.  How do you know there is t a problem with your device?

StefaanAvonds, 2017-02-21:
Because the problem occurs also with 2 other Android devices.

aritchie, 2017-02-21:
Please do not open duplicates of your own issue in the future.  

You had said in your previous defect that other apps were not picking up the other services (this library was finding only 1 service).

Now you are telling me it isn't finding any services or characteristics?  I have yet to reproduce this and you're giving me conflicting reports.

At this point, you'll need to send a reproducible sample and you'll need to add any additional info you can think of if you want this issue solved.

dramosti, 2017-02-22:
i have the same problemn !!  

BleAdapter.Current.Scan()  not return nothing !!

aritchie, 2017-02-22:
@dramosti this issue is not about scan.  Please do not hijack this thread.  

Secondly, Scan() has a known defect if you are filtering on serviceUUID.  There is an open defect for that.  If you simply need help, make sure you have read the documentation.  If you are still having an issue, please post a new defect.  

Lastly, please post additional details/useful information or do not post at all... thank you.

aritchie, 2017-03-02:
I have tested this with a bean+ without issue.  I have tried this with a Nexus 6 on android 7 and a google tab a on android 6 again the upcoming v2.

If you can provide additional information, please reopen/postback to this thread and I will take a look.

closed 2017-03-02 by aritchie

CJACQUEL, 2017-03-09:
Same problem: 
My code below : 
      protected override async void OnAppearing()
        {
            base.OnAppearing();

            var scanner = CrossBleAdapter.Current.Scan().Subscribe(async scanResult =>
            {
                 Byte[] bytes = new Byte[1024];

                ndevice = $"{scanResult.Device.Name}:{scanResult.Device.Uuid}:{scanResult.Rssi}";
                mDevices.Add(ndevice);
                if (scanResult.Device.Name == "TEST_2EB26A")
                {
                    m_device = scanResult.Device;
                    await m_device.Connect();
                    var serviceDiscover = m_device.WhenServiceDiscovered().Subscribe(service =>
                    {
                        var result = service;
                    });

                    m_device.WhenAnyCharacteristicDiscovered().Subscribe(characteristic =>
                    {
                        // read, write, or subscribe to notifications here
                        var result = characteristic.Read(); // use result.Data to see response
                                                            //   characteristic.Write(bytes);
                    });
                }
            });
      }
No service nor characteristic.
Thx for your help
Christophe

aritchie, 2017-03-09:
@CJACQUEL Please supply new info.  Perhaps a full test project with the issue?  Does the sample app work for you?  What's the platform and version you are using?

StefaanAvonds, 2017-03-09:
@CJACQUEL  What is your Android version? Have you tried adding the permissions to programmatically for Android (which solved the problem for me)?
```

        private readonly string[] Permissions =
        {
            Manifest.Permission.Bluetooth,
            Manifest.Permission.BluetoothAdmin,
            Manifest.Permission.BluetoothPrivileged,
            Manifest.Permission.AccessCoarseLocation,
            Manifest.Permission.AccessFineLocation
        };

        protected override void OnCreate (Bundle bundle)
        {
            base.OnCreate (bundle);
            Xamarin.Forms.Forms.Init(this, bundle);

            CheckPermissions();

            var app = new App();
            LoadApplication(app);
        }

        private void CheckPermissions()
        {
            bool minimumPermissionsGranted = true;

            foreach (string permission in Permissions)
            {
                if (CheckSelfPermission(permission) != Permission.Granted) minimumPermissionsGranted = false;
            }

            // If one of the minimum permissions aren't granted, we request them from the user
            if (!minimumPermissionsGranted) RequestPermissions(Permissions, 0);
        }
```

CJACQUEL, 2017-03-09:
No CrossMVM. I use Visual 2015, on Windows 7. The mobile is a galaxy tab 10. The USB device is made by our electronic lab. I will test the sample App.  Thx

CJACQUEL, 2017-03-09:
scanResult.Device has always the status disconnected even I use scanResult.Device.Connect();

StefaanAvonds, 2017-03-09:
This is how I do it:
```

device.Connect().Subscribe(o => { });
device.WhenStatusChanged().Subscribe(status =>
{
    // The actions...
});

```
If I'm not subscribing to the Connect() then the status indeed does not change.

aritchie, 2017-03-09:
await and subscribe will produce similar results.

CJACQUEL, 2017-03-10:
Hello, 
WhenStatusChanged is an answer to connect 
But WhenServiceDiscovered is corresponding to which command ? 
And WhenAnyDescriptorDiscovered, and WhenAnyCharacteristicDiscovered ? 
Thx, 
Christophe

aritchie, 2017-03-10:
Guys - this is a closed defect, not a chat room or support channel.  I'll answer this last question, but please move all future chats like this to xamarin forums and/or xamarin slack.

@CJACQUEL Please take some time to review the documentation.  If you don't really care about the services (for custom devices, you really don't need to be worried unless you have a lot of characteristics), you can use WhenAnyCharacteristicDiscovered().  This will return all characteristics across all services.  You likely don't need descriptors, though I don't know your product.  You should take a look at the bluetooth specification to see what descriptors are used for.

@StefaanAvonds I would appreciate it in the future if you would let me know that you've resolved your issue.  I spent a fair time looking into this to make sure things were working properly.  If you put your permissions into the manifest, you shouldn't need to request the permission programmatically.