aritchie/bluetoothle #18: Write and Wait for notification

oOGabsOo, 2017-02-21:
Hello,

Is there a way to Write on a Characteristic and Wait for the Notification in one command and get the result?

Thank you in advance for your help

aritchie, 2017-02-21:
Not from an API perspective.  You would have to do a write then read yourself.

oOGabsOo, 2017-02-21:
Thank you. I'm not really familiar with BLE, but the value that i'm expecting is the one from the notification and not the one that I can read myself after. Or both are the same? 

Do you think this is some features that you will consider to add in the API ?

aritchie, 2017-02-21:
I have no idea on how your BLE device works, so I can't really comment on your notification vs read scenario.  

Adding  something like this to the API wouldn't make sense.  Your write characteristic could be different from your read characteristic (and often is - most have a UART setup when cases like this arise) and it really isn't well suited for BLE to begin with.

closed 2017-02-21 by aritchie