aritchie/bluetoothle #22: Implementation

yiyoMX, 2017-02-25:
Hi, I have a UWP app, that connects via ble to a custom developed module. Im porting it to Xamarin, Im trying to implement your library, but Im not familiar with reactive programming. Is there any complete implementation example elsewhere? The one included in the readme isnt clear enough.
Thanks in advance!

aritchie, 2017-02-25:
No idea about other examples.  The sample I have is about as clear as it can get.  I do use some IoC concepts, but if you focus on the actual ble portion, it should be fairly straight forward.

yiyoMX, 2017-02-25:
Ok, I'll dive into it today. Great job by the way, thanks!

closed 2017-02-25 by yiyoMX

yiyoMX, 2017-02-25:
You are correct, the examples are very clear, I found that I was missing:

> using System.Reactive.Linq; 

Which translated in an error when trying to await an observable collection.
VS does not recommends it as a solution.