aritchie/bluetoothle #23: NUGET 2.0.0 STABLE NO VS REFERENCES

yiyoMX, 2017-03-07:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [x] PCL

## Version of OS(s) listed above with issue
VS NUGET

## Version of ACR BluetoothLE Library
2.0.0 STABLE

## Expected Behaviour
When upgrade or fresh installed versión 2.0.0 stable, the references arent added to the projects, with the beta 4 versión they do.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible
Contact GitHub

aritchie, 2017-03-07:
Looks fine to me.  You should be aware that the package and assembly name have changed from Acr.Ble to Plugin.BluetoothLE

yiyoMX, 2017-03-07:
Ok, I was seeing those but wrongly asumed that they where from another Nuget, sorry about that.
Also thanks for the plugin, and opensourcing the code!
Using it in Xamarin Forms.

closed 2017-03-07 by yiyoMX