aritchie/bluetoothle #25: Crash on AbstractDevice Finalize method

ajchacko, 2017-03-10:
Hello Allan,

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [] PCL

## Version of OS(s) listed above with issue
Android 6.0.1 (haven't tested with other versions but I assume it happens on all of them)

## Version of Library
2.0.0

## Expected Behaviour
When the finalize method gets called in AbstractDevice, it shouldn't crash the app.

## Actual Behavior
Currently in the Dispose method in Device (Plugin.BluetoothLE.Android), the following code causes a crash (called when the finalize method is called in AbstractDevice). I am able to replicate it in the sample app

```
public override void Dispose(bool disposing)
        {
            this.Context.Dispose(); //this is causing the crash
            base.Dispose(disposing);
            
        } 
```

`Unhandled Exception:
03-10 11:25:02.798 E/mono    (17674): System.ObjectDisposedException: Cannot access a disposed object.
03-10 11:25:02.798 E/mono    (17674): Object name: 'Android.Bluetooth.BluetoothDevice'.
03-10 11:25:02.798 E/mono    (17674):   at Java.Interop.JniPeerMembers.AssertSelf (Java.Interop.IJavaPeerable self) [0x00030] in /Users/builder/data/lanes/3511/501e63ce/source/Java.Interop/src/Java.Interop/Java.Interop/JniPeerMembers.cs:153 
03-10 11:25:02.798 E/mono    (17674):   at Java.Interop.JniPeerMembers+JniInstanceMethods.InvokeAbstractObjectMethod (System.String encodedMember, Java.Interop.IJavaPeerable self, Java.Interop.JniArgumentValue* parameters) [0x00002] in /Users/builder/data/lanes/3511/501e63ce/source/Java.Interop/src/Java.Interop/Java.Interop/JniPeerMembers.JniInstanceMethods_Invoke.cs:335 
03-10 11:25:02.798 E/mono    (17674):   at Android.Bluetooth.BluetoothDevice.ConnectGatt (Android.Content.Context context, System.Boolean autoConnect, Android.Bluetooth.BluetoothGattCallback callback) [0x00069] in /Users/builder/data/lanes/3511/501e63ce/source/monodroid/src/Mono.Android/platforms/android-24/src/generated/Android.Bluetooth.BluetoothDevice.cs:243 
03-10 11:25:02.798 E/mono    (17674):   at Plugin.BluetoothLE.Device.get_Context () [0x0000f] in C:\Dev\bluetoothle\Plugin.BluetoothLE.Android\Device.cs:48 
03-10 11:25:02.798 E/mono    (17674):   at Plugin.BluetoothLE.Device.Dispose (System.Boolean disposing) [0x00001] in C:\Dev\bluetoothle\Plugin.BluetoothLE.Android\Device.cs:447 
03-10 11:25:02.798 E/mono    (17674):   at Plugin.BluetoothLE.AbstractDevice.Finalize () [0x00002] in C:\Dev\bluetoothle\Plugin.BluetoothLE.Abstractions\AbstractDevice.cs:20 `

The device is already disposed I assume, so I am not sure if we need to dispose the context separately (which in turn calls Gatt.Close()).

Let me know if you need more info. Thanks a lot.

aritchie, 2017-03-10:
@ajchacko Keep an eye on the dreaded "GATT 133" error due to this being removed.  @MisterJimson 's fantastic Pixel suffers from this semi-rare condition.  Try the sample by thrashing the connect/disconnect to see if it blows up on his device.

aritchie, 2017-03-10:
ps - 2.0.1 is out on nuget for you

ajchacko, 2017-03-10:
Will do, thx a lot for the quick reply :)

ajchacko, 2017-03-10:
I tried this on a Samsung Tab E tablet and ran into the "GATT 133" error after a couple of connects/disconnects. I also tried it after uncommenting the code you had left there, but still ran into the issue. I think the issue is the dispose method is never called on AbstractDevice (it only disposes when the finalize method is called)

```
public void Dispose()
        {
            this.Dispose(true);
            GC.SuppressFinalize(this);
        }
```

aritchie, 2017-03-10:
I'm pretty sure this is why I was calling .Close() vs .Disconnect().  This fix is a bit more involved now.  I will look into it.

aritchie, 2017-03-13:
the branch "defect_25" has a potential fix.  I don't have a device reproducing this issue at the moment.

ajchacko, 2017-03-13:
I did some testing against the defect_25 branch and it looks like it resolves the issue. I am not getting the Gatt 133 error anymore and no more crash either :) thx

aritchie, 2017-03-13:
Thanks @ajchacko for the testing.  I'll package this up tonight and push it to nuget.

aritchie, 2017-03-13:
2.0.2 is out

commit d2135da9 (2017-03-10): fix #25 - finalizer for android.  leave comments for better fix to try later
commit 4686b45e (2017-03-17): fix #25 - always call Disconnect & Close.  Don't use dispose.  Let the GC cleanup
closed 2017-03-13 by aritchie