aritchie/bluetoothle #26: Ble reconnects

yiyoMX, 2017-03-12:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [x] PCL

## Version of OS(s) listed above with issue
Only tested in Xamarin Android (Forms Project).

## Version of Library
2.0.0 Stable

## Expected Behaviour
After a CancelConnection() the device must stay disconnected.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
After a CancelConnection() the device must stay disconnected. but it reconnects after a few seconds.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible
Contact GitHub

aritchie, 2017-03-13:
What version of android did you reproduce this on?
Did you call "CancelConnection" during a connection attempt?

yiyoMX, 2017-03-13:
Android 6, no I'm calling CancelConnection while already connected, does its supposed to work this way?

aritchie, 2017-03-13:
It is not supposed to work this way.  I did not reproduce this issue, but the reconnection logic can get complicated, so I've made a few additional checks as part of 2.0.2 which is on nuget now.  Please give that a try.  If there is still an issue, please report back in this issue.

yiyoMX, 2017-03-13:
I see the 2.0.2 now, upgrading!

yiyoMX, 2017-03-13:
Ive upgraded and now the first disconnection it stays disconnected, if I reconnect w/o closing the app, it connects, but If I try a second disconnection it stays connected forever. Gonna debug more, closing and reopening if necessary.
Thanks for your time.

closed 2017-03-13 by yiyoMX