aritchie/bluetoothle #27: Is there common code between BLEServer and BluetoothLE?

sledgehammer777, 2017-03-15:
Meaning, can I use both packages to find and get characteristics and then pass the characteristics to a common device processing class?  Our application is designed to work with devices that could be in peripheral or master mode depending on the customer's needs.  I'd like to be able to write code using each package to find and discover the devices depending on which mode the app is in and then pass an IDevice and 2 IGattCharacteristic objects to the common class for processing.  Is that possible?

aritchie, 2017-03-15:
There is no common code between them.  What you need to do isn't complicated though.  You will just have to do it in your logic in your app to translate between the two.

sledgehammer777, 2017-03-15:
That is what I thought.  Thanks for the quick reply!

closed 2017-03-15 by aritchie