aritchie/bluetoothle #28: Gatt Characteristic Write Callback Handler Question

ajchacko, 2017-03-15:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [] PCL

## Version of OS(s) listed above with issue
6.0

## Version of Library
2.0.2

## Expected Behavior
The writes/reads should work consistently

## Actual Behavior
I noticed some inconsistent behavior on Android when doing a write/read. The writes/reads work sometimes but not all the time, I think the following might be causing the issue. 

In the GattCharacteristic class for Android (in the Write, Read and SubscribeToNotifications methods) the handler for the CharacteristicChanged callback contains the following comparison 

```
if (args.Characteristic.Equals(this.native))
{
}
```
Is there a reason why the Uuid is not being used for this (like in iOS). I tried changing the code to the following and it seems to have fixed the issue. 

```
if (args.Characteristic.Uuid.Equals(this.native.Uuid))
{
}
```

Thanks

aritchie, 2017-03-17:
You don't want to compare UUIDs because you may have multiple connections to multiple devices of the same type.  You will be overlapping events between them.  The fact that I'm doing this on iOS is a bug.  I could run the compare all the way down to the service and device to make sure the instance is the same to trigger the event, but the instance comparison here should be enough.

Do the read/writes stop working outright or do you get the occasional timeout?

ajchacko, 2017-03-17:
Thanks Allan, I get timeouts every other call (read/write). When I debug, the args.Characteristic and this.native have different values (but the UUIDs are same). So there might need to be some other type of device/service level comparison like you suggested.

aritchie, 2017-03-17:
That is weird.  Same connection/session?  You aren't storing a reference to the characteristic anywhere are you?

ajchacko, 2017-03-17:
Its the same session, this is what I am doing. Not sure if this is helpful

```
{
    var vspTx = await device.GetCharacteristic(CharacteristicIds.VspTx).FirstAsync()
                .Timeout(TimeSpan.FromSeconds(timeOutInSeconds));

   await vspTx.Write(bytes)
                     .Timeout(TimeSpan.FromSeconds(timeOutInSeconds))
                     .RunAsync(cancelSrc.Token);

}
static IObservable<IGattCharacteristic> GetCharacteristic(this IDevice device, Guid characteristicUuid)
        {
            return device
               .WhenAnyCharacteristicDiscovered()
               .Timeout(TimeSpan.FromSeconds(timeOutInSeconds))
               .Where(x => x.Uuid.Equals(characteristicUuid));
        }
    }
```
The only way I can replicate this in the sample app is by connecting to two devices (of the same type I assume). After connecting to the 2nd device, if you go back and try to read a characteristic from the device it fails (args.Characteristic and this.native have different values). 

I am not sure if the following method in GattService on android is causing the issue or not (but this is the only place that is initializing the GattCharacteristic which changes the native value). 
```

public override IObservable<IGattCharacteristic> WhenCharacteristicDiscovered()
        {
            this.characteristicOb = this.characteristicOb ?? Observable.Create<IGattCharacteristic>(ob =>
            {
                foreach (var nch in this.native.Characteristics)
                {
                    var wrap = new GattCharacteristic(this, this.context, nch);
                    ob.OnNext(wrap);
                }
                return Disposable.Empty;
            })
            .Replay()
            .RefCount();

            return this.characteristicOb;
        }
```
I will keep looking into this and let you know if I find something useful.

Thanks

aritchie, 2017-03-17:
The important part is to make sure the services & characteristics aren't pinned across connections.  Normally, that replay would be very bad, however, the device clears out all of the services (characteristic parent) every disconnection - https://github.com/aritchie/bluetoothle/blob/master/Plugin.BluetoothLE.Android/Device.cs#L197

I could (should) switch this to every connect status.

ajchacko, 2017-03-21:
Hey Allan, I tried it with the change you made but am still running into this issue. Any other suggestions? I am running into this issue with just one connection.

aritchie, 2017-03-23:
Hey Ajay - I've been working on a fix based on your fix with the UUID.  The solution I have so far is checking all the way down to the device to ensure that the characteristics aren't "crossed".  I should be finished testing by the weekend.

aritchie, 2017-03-30:
2.0.3 was released a couple of days ago that should be fixed up for you.

closed 2017-03-30 by aritchie