aritchie/bluetoothle #29: GetKnownService/GetKnownCharacteristic

mariusbloemhof, 2017-03-29:
Hi, I am really liking your plugin so far, but there are a few things that is stopping me from using it. We have specialized firmware that requires authentication values to be written to the peripheral after connection otherwise it will terminate the connection - there is a timeout period for this.

We have very specific, known services and characteristics that is responsible for this. Hence it is important to us to be able to connect to a known service and known characteristic for this known service. Currently we have to check each characteristic/service as it is discovered to find the one were are looking at. Because of the limited time we have to send the authentication value, and the fact that having to scan for all services and all characteristics is too expensive, we reach the timeout before we can send the required values.

We are therefore looking for something like

IGattCharacteristic characteristic = device.**GetKnownService**(_UUID of service_)
    .**GetKnownCharacteristic**(_UUID of characteristic_); 

I know your time is precious, and this is not an issue, but a feature request, sorry I did not know where else to put it.

Thanks
Marius

mariusbloemhof, 2017-03-30:
It of course goes without saying that in the implementation, this should directly connect to the known services/characteristics and not just filter/linq to it :)

aritchie, 2017-04-13:
Done for next release

closed 2017-04-13 by aritchie