aritchie/bluetoothle #30: Java.Lang.IllegalArgumentException: {00000000-0000-0000-0000-987bf3767eea} is not a valid Bluetooth address

mariusbloemhof, 2017-03-29:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] Android

## Version of OS(s) listed above with issue

 - [x] Android

## Version of Library
2.03

## Expected Behaviour
            
var guid = new Guid("{00000000-0000-0000-0000-987bf3767eea}");
Device = CrossBleAdapter.Current.GetKnownDevice(guid);

I expect a device to be created so that I can connect to it

## Actual Behavior
 Java.Lang.IllegalArgumentException: {00000000-0000-0000-0000-987bf3767eea} is not a valid Bluetooth address

see attached stack trace

[stacktrace.txt](https://github.com/aritchie/bluetoothle/files/880451/stacktrace.txt)

aritchie, 2017-03-30:
I'm actually going to remove this method because it can't effectively be used

mariusbloemhof, 2017-03-30:
Hi aritchie,

Is it going te replaced with another similar mechanism? We really need the ability to connect to a known device, and not just being able to scan for devices that match a peripheral?

mariusbloemhof, 2017-03-30:
I haven't checked IOS, but this fixes the problem for Android

        public IDevice GetKnownDevice(Guid deviceId)
        {
            // TODO: guid flip the end in toString.... need to test this
            var native = this.manager.Adapter.GetRemoteDevice(deviceId.ToByteArray().Skip(10).Take(6).ToArray());
            var device = this.context.Devices.GetDevice(native, TaskScheduler.Current);
            return device;
        }

aritchie, 2017-03-31:
I'll take a look and see what can be done.  iOS was my worry for this one.

closed 2017-04-13 by aritchie