aritchie/bluetoothle #31: System.FormatException.Unrecognized Guid format

v9rt3x, 2017-04-03:
- [] iOS
 - [x] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [] PCL

## Version of OS(s) listed above with issue
6.0.1

## Version of Library
2.0.3

## Expected Behaviour
Getting from scan result object the advertisement data.

## Actual Behavior
What I'm getting back:
-objScanResonse
--AdvertisementData
---IsConnectable: System.FormatException.Unrecognized Guid format.
---LocalName: System.FormatException.Unrecognized Guid format.
---ManufacturerData:System.FormatException.Unrecognized Guid format.
..
--Device: {Device: 00000000-0000-0000-0000-d03972c4b51a}
---Rssi: -49
..

Thanks in advance for help.

aritchie, 2017-04-03:
Please include the advertised service UUIDs of the device

v9rt3x, 2017-04-03:
Thats the point. I can't read the AdvertisementData object.

If I do s.th. like:
..
```
.Scan()
.Subscribe(objScanResponse =>{
foreach (Guid objGuid in objScanResponse.AdvertisementData.ServiceUuids)
{
	 Debug.WriteLine(objGuid.ToString());
}

```
The Error is then for the Line "objScanResponse.AdvertisementData.ServiceUuids":
System.FormatException has been thrown.
Unrecognized Guid format.

aritchie, 2017-04-03:
I get that.  Please retrieve them using another app.  I haven't observed this on the devices I have so I need a reproducible sample of some sort.  I do seem to be accounting for the different types of UUIDs that should be seen.

v9rt3x, 2017-04-03:
With another App I get:

**Raw AD Records:**

**#1 Flags:**
As Array: '[05]'

**#2 More 16-bit UUIDs available**
As Array: '[03, 18, 02, 18, 04, 18]'

**#3 Name (Complete)**
As Array: '[xx, xx, xx, xx........]'

aritchie, 2017-04-21:
I can't see why this is a problem and it is passing my tests.  I have added a way to at least prevent it from throwing an error, but when the condition is detected, I return a "dummy" GUID.

Please try the v3.0 branch and run a scan to see if it alleviates this issue.

aritchie, 2017-04-29:
I haven't heard back, so I'm going to assume this is closed.

closed 2017-04-29 by aritchie