aritchie/bluetoothle #32: Only able to subscribe to one characteristics notification

tplooker, 2017-04-13:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [] PCL

## Version of OS(s) listed above with issue
6.0

## Version of Library
2.0.3

## Expected Behaviour
Able to subscribe to multiple BLE notifications

## Actual Behavior
I am subscribing to the notifications as they are discovered (kind of like in your example code on the main page) however I only ever receive notifications from the first characteristic, It is a custom GATT service coming from an Adafruit bluefruit module, I have confirmed through other apps that all the characteristics are infact sending notifications. Using your example app and selecting the characteristics manually for notification works as expected but for some reason subscribing on discovery doesn't work. Are there any time constraints you are aware of like minimum time from connection or the last subscription event that I need to wait for?

Thanks!

aritchie, 2017-04-13:
I use this type of onconnect/subscribe functionality in a project right now without issue.  The one thing I find is that on android, if your characteristic is broadcasting data too often (broadcast more than one a second) is that it will flood the pipe on android.  It also tends to cause timeouts on any writes you may be doing.  

If this is not the case, please send me a reproducible sample that I can take a look at.

tplooker, 2017-04-13:
Thanks for the quick response so I have a class called BLEDevice that has the following constructor.

public BLEDevice(IDevice device)
{
        Device = device;

        Device.WhenServiceDiscovered()
                   .Subscribe(service =>
                   {
                          setService(service);
                   });
}

Where the setService is defined as the following.
private async void setService(IGattService service)
 {
           if (service.Uuid.ToString() == DeviceInfoGattService.ServiceUuid.)
            {
                DeviceInfoService = new DeviceInfoGattService(service);
            }
 }

The deviceinfogattservice class extends the abstract class I have made called AbstractGattService which calls the following constructor.

protected AbstractGattService(IGattService gattService)
{
            _gattService = gattService;
            _gattService.WhenCharacteristicDiscovered()
                .Subscribe(characteristic =>
                {
                    setCharacteristic(characteristic);
                });
}

where setCharacteristic is an abstract method implemented by each service that extends AbstractGattService, in the case of the DeviceInfoService, the setCharacteristic() method has the following definition.

 protected override void setCharacteristic(IGattCharacteristic characteristic)
{
            if (characteristic.Uuid == _wifiStatusUuid)
                _wifiGattCharacteristic = new GattCharacteristic(characteristic);
            else if (characteristic.Uuid == _loggingStatusUuid)
                _loggingGattCharacteristic = new GattCharacteristic(characteristic);
            else if (characteristic.Uuid == _sensingStatusUuid)
                _sensingStatusGattCharacteristic = new GattCharacteristic(characteristic);
}

Where the relivant methods from the GattCharacteristic class are the following.

private IGattCharacteristic _characteristic;
public IGattDescriptor Descriptor { get; set; }

public GattCharacteristic(IGattCharacteristic characteristic)
{
            _characteristic = characteristic;
            if (_characteristic.CanNotify())
            {
                _characteristic.SubscribeToNotifications()
                    .Subscribe(result => { handleRecievedNotification(result); });
            }
            _characteristic.WhenDescriptorDiscovered()
                .Subscribe(descriptor =>
                {
                    Descriptor = descriptor;
                });
}

public void handleRecievedNotification(CharacteristicResult result)
{
            Data = result.Data;
}

If I step through the code until the last time the setCharacteristic() method is called by the DeviceInfoService class and debug the characteristic object for the wifistatuscharacteristic I see the IsNotifying property is set to true but for the other two characteristics (loggingGattCharacteristic ,sensingStatusGattCharacteristic) that have been through the exact same subscription to notifications process, the IsNotifying property is set to false. Consequently when these two latter characteristics (loggingGattCharacteristic ,sensingStatusGattCharacteristic) are changed the handleRecievedNotification event does not fire but it does for the wifistatuscharacteristic. What is strange is when I subscribe to these three notifications using your example app, I get no problems. It is obviously an implementation problem on my behalf but I cannot figure out what I have done wrong.

Thanks for your help!
Tobias

aritchie, 2017-04-14:
First big thing is to store you subscription to a variable to make sure the GC doesn't clean it up.

instanceVar = _characteristic.SubscribeToNotifications()
                    .Subscribe(result => { handleRecievedNotification(result); });

tplooker, 2017-04-14:
So I tried this earlier as I saw it in your example but in my case it didn't make any difference, the notification watcher for these three characteristics is set as expected too.
```
        private IGattCharacteristic _characteristic;
        private IDisposable _notifcationwatcher;

        public GattCharacteristic(IGattCharacteristic characteristic)
        {
            _characteristic = characteristic;
            if (_characteristic.CanNotify())
            {
                _notifcationwatcher = _characteristic.SubscribeToNotifications()
                    .Subscribe(result => { handleRecievedNotification(result); });
            }
            _characteristic.WhenDescriptorDiscovered()
                .Subscribe(descriptor =>
                {
                    Descriptor = descriptor;
                });
        }
```

Thanks!

aritchie, 2017-04-18:
I can't see anything wrong with the code you've shown.  I'm afraid you'll have to send a small reproducible sample if you want me to take a look.

aritchie, 2017-04-19:
I have discovered a situation (a reproducible device) where this issue occurs.  You can overwhelm the system with writes.  Android doesn't queue the read/write calls very well.  The ones that fail, fail silently.  

I will attempt to sync up the requests which should fix this problem

tplooker, 2017-04-19:
Great, thanks for taking a look for me!

Tobias

aritchie, 2017-04-21:
Please try 3.0.0-beta1 on nuget to see if this resolves your issue.  I have done some very preliminary testing and it seems alright.

tplooker, 2017-04-21:
Thanks Allan I will try this in the next few days, appreciate your help!

tplooker, 2017-04-24:
Hi Allan,

Unfortunately this hasn't cured my issue, do you want me to send you a reproducible sample of my code?

Thanks
Tobias

aritchie, 2017-04-24:
Hi @tplooker 
Thanks for the ping back.  I definitely know what the issue is, solving it is the tough part.  If you have time, please submit your code.  There is a good chance I won't need it, but at least I can look if necessary.  The test case itself (a remote device + a phone that has the issue) is really my key issue at the moment.

aritchie, 2017-05-07:
v3-beta3 seems to finally fix this issue against my reproducible sample.  Basically, any read/write must be synchronized over to the device.  Please give it a try and let me know

aritchie, 2017-05-23:
For those still struggling with this issue, if you force the characteristics to register in-sync, everything will work.
```csharp
                                service
                                    .WhenCharacteristicDiscovered()
                                    .Take(5)
                                    .Subscribe(x =>
                                    {
                                        lock (sync)
                                        {
                                            x.SubscribeToNotifications().Subscribe(this.OnCharacteristicNotification);
                                            Task.Delay(TimeSpan.FromSeconds(1)).Wait();
                                        }
                                    });
```

aritchie, 2017-10-24:
As of 5.1-beta2, this is considered case closed

closed 2017-10-24 by aritchie