aritchie/bluetoothle #33: Expose native device property

aalmada, 2017-04-14:
I'm working on a Xamarin.Forms app that uses your plugin to scan for devices but want to use the Bluetooth device's SDK to connect and handle communications. They supply a separate SDK for each mobile platform. I thought about porting it to C# based on your plugin but it looks way too complex.
To be able to use their SDK I need access to the native device class instance, for example, the Android's BluetoothDevice class.
I can cast IDevice to the platform specific implementation but the native device class is private. I'd like not to have to resort to reflection to access it. Is it possible you make these public? What's you opinion of this?
I can try to make the change and send a PR. I just don't know which is the best branch to do this on.
Thanks in advance.

aritchie, 2017-04-14:
I'll expose IDevicer.NativeDevice (object get) in the next release

aalmada, 2017-04-15:
Cool! Thanks...

aritchie, 2017-04-21:
This is done for next major version

closed 2017-04-21 by aritchie