aritchie/bluetoothle #34: Scanning throw "Your adapter status is Unknown" error on iOS.

SuhanKoh, 2017-04-19:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [Y] iOS
 - [] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [Y] PCL

## Version of OS(s) listed above with issue
iOS 10+ iPhone 6s plus

## Version of Library
2.0.2

## Expected Behaviour
When you call 
```
            CrossBleAdapter.Current.Scan().Subscribe(scanResult =>
            {
                Debug.WriteLine("Found : " + scanResult.Device.Name);
            });
```
It supposed to return a list of found devices, but on iOS.

## Actual Behavior
Thrown "System.ArgumentException: Your adapter status is Unknown"

aritchie, 2017-04-19:
Check your adapter status before scanning.  There is an WhenStatusChanged() observable as well as a Status property.

Also, you shouldn't scan the second your app starts as the BLE state may not have been detected yet.

aritchie, 2017-04-19:
And also make sure you've given permissions to your app.  Follow the instructions in the readme

closed 2017-04-19 by aritchie

SuhanKoh, 2017-04-19:
I am using ScanWhenAdapterReady instead of Scan now.
Should this method documented?

aritchie, 2017-04-19:
all methods should be documented, I just don't have the time in life right now.

In any case, that method will help solve your issue, but you really should check the status observable as well so at least your user knows what is going on.

SuhanKoh, 2017-04-19:
Thanks, will do.