aritchie/bluetoothle #35: Expose Advertisement Service Data

HenrySipp, 2017-04-20:
I'm working on a forms app for mobile devices and if possible I would love to use this library as a way to communicate with our client's eddystone beacons. Unfortunately, a large part of the eddystone spec is read through the Advertisement Service data, can this be exposed?

aritchie, 2017-04-21:
Yes.  It will be available in the next major release

closed 2017-04-21 by aritchie

aritchie, 2017-04-24:
This is available as part of 3.0.0-beta1 on nuget.  Please give it a try and let me know your results.

HenrySipp, 2017-04-24:
From preliminary testing, this works wonderfully.

bodeg, 2019-01-29:
I have Plugin.BluetoothLE 6.2.3 installed. Thanks for that great library.

I am using the library to discover different BLE devices.
When discovering beacons:
On Android, the Advertisement Service data is not available.
On iOS, both the Advertisement Service data and the Manufacturer Data are not available

 _adapter
                .Scan()
                .Buffer(TimeSpan.FromSeconds(1))
                .Subscribe(scanResults => ...);

any idea what I am doing wrong?
With regards. alon.

aritchie, 2019-01-29:
@bodeg please don't file things in closed issues.  

You actually need access to the manufacturer data which works fine on android for me.

However, on iOS, beacon manufacturer data is hijacked by the OS because they have an ibeacon mechanism they want you to use.