aritchie/bluetoothle #36: Reconnection (IsPersistent or manual reconnection) fails on characteristics

mariusbloemhof, 2017-04-25:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS

## Version of OS(s) listed above with issue
8.2

## Version of Library

current master from git (tried branch 3.0 as well)

## Behaviour

upon disconnection of a device, I start to scan to re-establish the connection. the connection is re-established, but accessing characteristics fails thereafter. 

using IsPersistent does this for me, but the result is the same.

Upon reconnection, this.Native.Peripheral is null,  (see screenshot)

mariusbloemhof, 2017-04-25:
`        public void ConnectDevice()
        {
            var scanSettings = new ScanConfig
            {
                ServiceUuid = Guid.Parse("{00001803-0000-1000-8000-00805f9b34fb}"),
                ScanType = BleScanType.Balanced,
            };
            scanner = CrossBleAdapter.Current
                .Scan(scanSettings)
                .Subscribe(async result => 
                {
                    if (result.Device.Name == "SensorTag" || result.Device.Name == "Shadow Device")
                    {
                        scanner?.Dispose();
                        var connectionConfig = new GattConnectionConfig
                        {
                            IsPersistent = true,
                            Priority = ConnectionPriority.Normal,
                            NotifyOnConnect = true,
                            NotifyOnDisconnect = true,
                            NotifyOnNotification = true,
                        };
                        await result.Device.Connect(connectionConfig);
                    }
                });

            CrossBleAdapter.Current.WhenDeviceStatusChanged().Subscribe(device => 
            {
                if (device.Status == ConnectionStatus.Connected)
                {
                    device
                        .WhenServiceDiscovered()
                        .Where(s => s.Uuid == Guid.Parse("0000ffe0-0000-1000-8000-00805f9b34fb"))
                        .Subscribe(service =>
                        {
                            service.WhenCharacteristicDiscovered()
                        .Where(c => c.Uuid == Guid.Parse("0000ffe1-0000-1000-8000-00805f9b34fb"))
                        .Subscribe(async characteristic =>
                                {
                                    if (characteristic.Uuid == Guid.Parse("0000ffe1-0000-1000-8000-00805f9b34fb"))
                                    {
                                        authorized = await BLEDeviceAuthorized(characteristic);
                                        if (authorized)
                                        {
                                            Debug.WriteLine("connected and authorized");
                                            await characteristic.Write(BLEConstants.STORED_KEYPRESS_ON);
                                            var notify = new BLENotify(characteristic);
                                        }
                                    }
                                });
                        });
                }
            });

        }

[image]
`

aritchie, 2017-04-25:
Please copy/paste the actual stacktrace in the future, not screenshots.

aritchie, 2017-04-30:
I found the NRE, but not in the same spot you found it.  In fact, the place where I found it is a known issue with ios.  Since I'm working on the 3.0 branch right now, please give 3.0-beta2 on nuget a try.  If you find it fixed there, I'll happily back port the fix into 2.0

mariusbloemhof, 2017-05-01:
Thanks, Allen, will try tonight and let you know

SoyPan, 2017-05-01:
Hi, I got the same error with the 3.0 beta 2 nugget as well. It happens when I try to find services after the reconnection; it works perfectly on the first connection:
```
public void DeviceStatusChanged(ConnectionStatus connStat)//object sender, DeviceConnectionEventArgs e)
        {
            switch (connStat)
            {
                case ConnectionStatus.Connecting:
                    matchingServices = 0;
                    break;

                case ConnectionStatus.Disconnected:
                    DiscoveredServices.Clear();
                    setToNotify = false;
                    ConnectedDevice = null;
                    connStatus = "disconnected";
                    break;

                case ConnectionStatus.Connected:
                    connStatus = "connected";
                    ConnectedDevice.WhenServiceDiscovered().Subscribe(ServiceDiscovered); //crashes here on reconnect only
                    break;
            }
        }
```

aritchie, 2017-05-01:
@SoyPan Are you sure the exception is identical and that you are using 3.0-beta2 (it only came out last night).  Please include the full stack trace in the future.

mariusbloemhof, 2017-05-01:
Hi, I am still getting the same error - will attach the stacktrace, call stack and all other debug info asap

SoyPan, 2017-05-02:
@aritchie I am really sorry, I had clicked update on my nugget but apparently it was failing on my iOS project for some reason and I didn't realize it. I uninstalled and reinstalled, and the problem went away. Thank you!

aritchie, 2017-05-02:
@SoyPan thanks for letting me know.

@mariusbloemhof thanks for the update.  I'll do some digging.  I had a slightly different issue with an NRE that I stopped by checking if the native properties were set.  I'm not sure why this isn't working for you.

mariusbloemhof, 2017-05-03:
@aritchie sure no problem - I will also see if I can isolate the problem.

It is definitely the native.Service which is the problem. It is weird, because if I put a breakpoint on the constructor where native gets assigned, native.Service is actually set. Somewhere between the constructor being called and SubscribeToNotifications, native.Service is lost.

I will keep on working on the 3.0 branch on the git repo

aritchie, 2017-05-05:
@mariusbloemhof I have been running the connect/disconnect scenario over and over without issue.  I need a couple of bits of info here
- What version of iOS are you running this on?  
- What version/type of device are you running?
- if (!this.native.Peripheral.Identifier.Equals(service.Peripheral.Identifier))  <= In this line, what variable is null?  I'm guessing this.native.Peripheral?

mariusbloemhof, 2017-05-07:
Hi @aritchie ,
 
It is definitely Native.Service that is null in class **GattCharacteristic**

This specific phone is IOS 8.2. I will test on a IOS 10 phone to see if there is any difference.

The device is a very specific custom device with custom firmware. However, we have full control over it, and every last behaviour is known and intended. 

I seem to have been able to reproduce the same behaviour by switching the radio on and off, as well as resetting the device. The points of failure are slightly different:

When I reset the device, it fails on GattCharacteristic.cs:131 (see stacktrace01): this.native.Service is null

When I switch the radio on and off, it fails on GattCharacteristic.cs:36 (see stacktrace02): this.native.Service is null

[stacktrace01(device reset).txt](https://github.com/aritchie/bluetoothle/files/981383/stacktrace01.device.reset.txt)
[stacktrace02(radio on_off).txt](https://github.com/aritchie/bluetoothle/files/981384/stacktrace02.radio.on_off.txt)

aritchie, 2017-05-07:
@mariusbloemhof thanks for the complete info.  It is extremely helpful.  I don't think this is a problem with your device at all.  The null parent objects is either a bug with xamarin or a bug iOS as some sort of race condition that I'm not compensating for.  I've got a good idea how to fix this now.

aritchie, 2017-05-07:
@mariusbloemhof I've refactored how the parent object are passed around in v3-beta3 (on nuget now) instead of relying on them being set internally.  This has to fix the issue :)

mariusbloemhof, 2017-05-09:
@aritchie, so far the issue seem to be fixed. I could not get the IsPersistent option to work, but when I manually initiate scanning on disconnection, it works. I will test some more and let you know

aritchie, 2017-05-09:
@mariusbloemhof excellent.  thanks for the update

The persistence logic hasn't changed and for the most part, I get very little say from iOS for how that works.  If the NRE issues are gone, we'll move on to another issue if necessary.

closed 2017-05-12 by aritchie