aritchie/bluetoothle #39: Correct way to write multiple value types?

HenrySipp, 2017-05-04:
I'm sorry if this isn't the right place to ask this question, but it's the best I could think of.

I'm trying to communicate with a BLE device that expects a sort of packet structure for its certain GATT characteristic. It expects a packet containing 3 unsigned bytes and one value of UInt16. I'm having difficulty writing this packet to the object as the primary GattCharacteristic.Write() call only accepts standard bytes.

aritchie, 2017-05-04:
Bytes are bytes.  Write is doing what the interface needs.  This is definitely a question you should be googling, but this is a pretty quick answer:

```csharp
var bytes = new byte[5];
bytes[0] = 1st byte;
bytes[1] = 2nd byte;
bytes[2] = 3rd byte;

var i = BitConverter.GetBytes(your ushort value);
bytes[3] = i[0];
bytes[4] = i[1];
```
Not the prettiest, but it will do the trick

closed 2017-05-04 by aritchie

HenrySipp, 2017-05-04:
Right, I thought this was just as easy as that but it's not working.
It must be an issue with how the device itself is expecting the data - I'll contact our device manufacturer on that information.

Sorry, I just assumed I was doing something wrong.