aritchie/bluetoothle #40: UWP Ready

vbisbest, 2017-05-07:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [] PCL

## Version of OS(s) listed above with issue

## Version of Library

## Expected Behaviour

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible
Contact GitHub

aritchie, 2017-05-07:
It would be advertised in the readme if it was

closed 2017-05-07 by aritchie

chandrakantnitt, 2017-05-07:
Sorry to bug you @aritchie but do you have an ETA for this?

aritchie, 2017-05-07:
@chandrakantnitt That depends... how's your PR for it coming?