aritchie/bluetoothle #41: Error when pairing

wmcainsh, 2017-05-08:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [x] PCL

## Version of OS(s) listed above with issue
iPhone 5c - 10.3.1

## Version of Library
2.0.3

## Expected Behaviour
Pair to device

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
`device.Features.HasFlag(DeviceFeatures.PairingRequests)` = true
`device.IsPairingAvailable()` = true

`device.PairingRequest().ToTask()` results in **System.ArgumentException: Pairing request is not supported on this platform**

### Stack Trace
  at Plugin.BluetoothLE.AbstractDevice.PairingRequest (System.String pin) [0x00000] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Abstractions\AbstractDevice.cs:40 
  at My.Mobile.ViewModels.DeviceViewModel+<OnPairToDevice>d__41.MoveNext () [0x000c9] in D:\Dev\VS2017\Projects\My.Mobile\My.Mobile\My.Mobile\ViewModels\DeviceViewModel.cs:117 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.8.0.175/src/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
  at System.Runtime.CompilerServices.AsyncMethodBuilderCore.<ThrowAsync>m__0 (System.Object state) [0x00000] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.8.0.175/src/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/AsyncMethodBuilder.cs:1018 
  at UIKit.UIKitSynchronizationContext+<Post>c__AnonStorey0.<>m__0 () [0x00000] in /Users/builder/data/lanes/4466/a04678c2/source/xamarin-macios/src/UIKit/UIKitSynchronizationContext.cs:24 
  at Foundation.NSAsyncActionDispatcher.Apply () [0x00000] in /Users/builder/data/lanes/4466/a04678c2/source/xamarin-macios/src/Foundation/NSAction.cs:163 
  at (wrapper managed-to-native) UIKit.UIApplication:UIApplicationMain (int,string[],intptr,intptr)
  at UIKit.UIApplication.Main (System.String[] args, System.IntPtr principal, System.IntPtr delegate) [0x00005] in /Users/builder/data/lanes/4466/a04678c2/source/xamarin-macios/src/UIKit/UIApplication.cs:79 
  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) [0x00038] in /Users/builder/data/lanes/4466/a04678c2/source/xamarin-macios/src/UIKit/UIApplication.cs:63 

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible
Contact GitHub

closed 2017-05-08 by aritchie

wmcainsh, 2017-05-08:
Does this mean that pairing from an iPhone isn't available?

aritchie, 2017-05-08:
that is correct

wmcainsh, 2017-05-08:
It might be worthwhile removing that as a supported feature from the main page then.

aritchie, 2017-05-08:
It works fine for android & eventually UWP.  I use it in my projects.

aritchie, 2017-05-08:
Plus - iOS does initiate the pairing action when you take any action against a secured characteristic.  You just have no control over it from an API perspective.

wmcainsh, 2017-05-08:
Unfortunately I need the pairing to happen automatically as the user isn't aware of the key.

aritchie, 2017-05-08:
Sorry - not much I can do to help you here.  It is an apple api (or lack thereof) problem.

wmcainsh, 2017-05-08:
No worries. I thought everything was going too smoothly..

commit 2fda4894 (2017-05-08): fix #41 - should allow MTU requests on iOS, not pairing