aritchie/bluetoothle #42: Reoppened #26: Reconnection after CancelConncetion()

v9rt3x, 2017-05-08:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [] PCL

## Version of OS(s) listed above with issue
6.0.1

## Version of Library
2.0.3

## Expected Behaviour
No reconnection after disconnect (CancelConnection())

## Actual Behavior
Sometimes the disconnect works and the device stays disconnected. But mostly the second time I try to disconnect after a connection the device immediately reconnects.

[BluetoothGatt] cancelOpen() - device: D0:5F:B8:5F:1D:39
[BluetoothGatt] close()
[BluetoothGatt] unregisterApp() - mClientIf=5
[BluetoothGatt] connect() - device: D0:5F:B8:5F:1D:39, auto: false
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=0ebbc9b3-8aac-4d56-9e97-be44665086d2
[BluetoothGatt] onClientRegistered() - status=0 clientIf=5

Thanks in advance!

aritchie, 2017-05-31:
I have no witnessed this problem.  Can you please provide sample code that shows the issue.

aritchie, 2017-06-06:
No response to request for info

closed 2017-06-06 by aritchie