aritchie/bluetoothle #43: Device reconnect error

wmcainsh, 2017-05-09:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] UWP
 - [] tvOS
 - [] macOS
 - [x] PCL

## Version of OS(s) listed above with issue
iPhone 5c - 10.3.1

## Version of Library
2.0.3

## Expected Behaviour
Device connects and retrieves services / characteristics
Device disconnects
Device re-connects and retrieves services / characteristics

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

Device connects and retrieves services / characteristics
Device disconnects
Device re-connects and throws error at:
[https://github.com/aritchie/bluetoothle/blob/master/Plugin.BluetoothLE.Apple.Shared/GattService.cs#L31](url)

`foreach (var nch in this.native.Characteristics)` this.native.Characteristics is null

#### Stack trace
  at Plugin.BluetoothLE.GattService+<>c__DisplayClass3_0.<WhenCharacteristicDiscovered>b__1 (System.Object sender, CoreBluetooth.CBServiceEventArgs args) [0x00089] in C:\Users\me\Downloads\bluetoothle-master\Plugin.BluetoothLE.Apple.Shared\GattService.cs:31 
  at (wrapper delegate-invoke) System.EventHandler`1[CoreBluetooth.CBServiceEventArgs]:invoke_void_object_TEventArgs (object,CoreBluetooth.CBServiceEventArgs)
  at CoreBluetooth.CBPeripheral+_CBPeripheralDelegate.DiscoveredCharacteristic (CoreBluetooth.CBPeripheral peripheral, CoreBluetooth.CBService service, Foundation.NSError error) [0x00015] in /Users/builder/data/lanes/4466/a04678c2/source/xamarin-macios/src/build/ios/native/CoreBluetooth/CBPeripheral.g.cs:402 

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible
Contact GitHub

wmcainsh, 2017-05-09:
Possibly a duplicate of #36, close this if it is.

aritchie, 2017-05-09:
This is indeed a duplicate that appears to be fixed in v3-beta3 which is on nuget

closed 2017-05-09 by aritchie