aritchie/bluetoothle #46: GetKnownCharacteristics() hangs on initially retrieved value

cpboyd, 2017-05-31:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [*] iOS

## Version of OS(s) listed above with issue
Tested on iOS 10.3.2

## Version of Library
3.1.1

## Expected Behaviour
I should be able to request and retrieve multiple characteristics, one after another, without issue.

## Actual Behavior
```
            currentDevice = (await ScanForDevices().FirstAsync(result => result.Device.Uuid == uuid)).Device;
            await currentDevice.Connect();
            var service = await currentDevice.GetKnownService(DeviceService);
            var rxChar = await service.GetKnownCharacteristics(DeviceRx).FirstAsync();
            var txChar = await service.GetKnownCharacteristics(DeviceTx).FirstAsync();
            var notif = txChar.SubscribeToNotifications();
```

This results in both `rxChar` and `txChar` containing **IDENTICAL** characteristics (corresponding to `DeviceService`/`DeviceRx`), even though the specified UUID is different.

As a result, `SubscribeToNotifications()` throws an error because `DeviceRx` only allows writes, thus providing no notifications.

If I comment out the `var rxChar` line, then `txChar` represents the proper characteristic and no error gets thrown.

cpboyd, 2017-05-31:
Workarounds:
```c#
var temp = await currentDevice.GetKnownCharacteristics(DeviceService, DeviceRx, FlirTx).ToList();
            foreach (var x in temp) {
				if (x.Uuid == DeviceRx)
				{
					rxChar = x;
				}
				else if (x.Uuid == FlirTx)
				{
					var notif = x.SubscribeToNotifications();
					if (onConnect != null)
					{
						notif = notif.Do(_ => onConnect());
					}
					notif.Subscribe(result =>
					{
						var str = Encoding.ASCII.GetString(result.Data);
						Console.WriteLine(str);
						responseSubject.OnNext(str);
					}).DisposeWith(subList);
				}
            }
```

or

```c#
            currentDevice.GetKnownCharacteristics(DeviceService, DeviceRx, FlirTx).Subscribe(x => {
				if (x.Uuid == DeviceRx)
				{
					rxChar = x;
				}
				else if (x.Uuid == FlirTx)
				{
					var notif = x.SubscribeToNotifications();
					if (onConnect != null)
					{
						notif = notif.Do(_ => onConnect());
					}
					notif.Subscribe(result =>
					{
						var str = Encoding.ASCII.GetString(result.Data);
						Console.WriteLine(str);
						responseSubject.OnNext(str);
					}).DisposeWith(subList);
				}
            }).DisposeWith(subList);
```

aritchie, 2017-05-31:
That isn't really a workaround.  That is how I intended it to work.

It should be able to handle consecutive calls.  I'll look into this for a future release.

aritchie, 2017-10-29:
no feedback and cant reproduce - closing

closed 2017-10-29 by aritchie