aritchie/bluetoothle #47: Suggestion: Way to tell if SubscribeToNotifications() is successful?

cpboyd, 2017-05-31:
I'd like to be able to trigger writes to a characteristic after I successfully subscribe to another characteristic's notifications.

On iOS, I've previously been using RxBluetoothKit which provides the ability to do this via [`setNotifyValue(_:)`](http://cocoadocs.org/docsets/RxBluetoothKit/3.0.6/Classes/Characteristic.html#/s:FC14RxBluetoothKit14Characteristic14setNotifyValueFSbGC7RxSwift10ObservableS0__)

Example:
```swift
return service.discoverCharacteristics([txChar])
            .flatMap { Observable.from($0) }.take(1)
            .flatMap {
                Observable.of(
                    $0.monitorValueUpdate(),
                    $0.setNotifyValue(true)
                        .do(onNext: {(_) in self.onNotificationsEnabled() })
                        .ignoreElements()
                        .subscribeOn(CurrentThreadScheduler.instance))
                .merge()
            }
```

On Android, I've been using RxAndroidBle which provides [`setupNotification()`](https://github.com/Polidea/RxAndroidBle#change-notifications)

Example (from URL):
```java
device.establishConnection(false)
    .flatMap(rxBleConnection -> rxBleConnection.setupNotification(characteristicUuid))
    .doOnNext(notificationObservable -> {
        // Notification has been set up
    })
    .flatMap(notificationObservable -> notificationObservable) // <-- Notification has been set up, now observe value changes.
    .subscribe(
        bytes -> {
            // Given characteristic has been changes, here is the value.
        },
        throwable -> {
            // Handle an error here.
        }
    );
```

Currently, `SubscribeToNotifications()` simply seems to enable notifications and then start subscribing.

Since the characteristic that I'm subscribing to is supposed to return data depending on commands that I send to another characteristic, I'd like to ensure that notifications are properly enabled before I start sending any commands.

aritchie, 2017-06-01:
This is already on my todo list for 3.2 or 3.3

aritchie, 2017-06-28:
This has been released with v4 - on nuget now

Use: characteristic.SetNotificationValue(CharacteristicConfigDescriptorValue.Notify); // or Indicate

closed 2017-06-28 by aritchie