aritchie/bluetoothle #51: Tx power always 0

Binjaaa, 2017-06-25:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [x] PCL
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android 6.0 - API 23 - Nexus 5

## Version of Library
3.1.1

## Expected Behaviour
TxPower is not 0

## Actual Behavior
After setting TxPower to -4 on my iBKS beacon (Accent-Systems) on a URL advertisement, I see 0 via `ScanResult.AdvertisementData.TxPower`

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible
Contact GitHub

aritchie, 2017-06-26:
This value is becoming increasingly blocked by APIs on both Android & iOS due to the implications with tracking a potential user.  They believe in the beacon which forces you to play by certain rules for tracking.

That being said, I wouldn't rely on this value at all and I'm likely to remove it in a future version.

aritchie, 2017-07-24:
I could not see this problem on Android, but on iOS this makes sense since beacon packets have their txpower intentionally blocked by iOS.

closed 2017-07-24 by aritchie