aritchie/bluetoothle #54: Characteristics on Multiple Devices.

SeppyMoose, 2017-06-29:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)
 - Android

## Version of OS(s) listed above with issue
5.1

## Version of Library
4.0

## Expected Behavior
Subscribing to characteristics from multiple devices. Receive notifications (indicate) from both.

## Actual Behavior
Currently the last subscribed device is the only one to send notifications.

##Minimum code:

//Starts the process
            CrossBleAdapter.Current.ScanInterval(new TimeSpan(0, 0, 10));
            ccss = CrossBleAdapter.Current.Scan().Subscribe(ScanResult => 
            {
                DeviceSubscriber_Added(ScanResult);
            });

//When a device is selected this is run, it does
            wsd = mDevice.WhenServiceDiscovered().Subscribe(
            obj =>
            {
                mServices.Add(new ServiceBLE(obj, ServiceReady));
            });

//When a service is added this is run
            wcd = pInput.WhenCharacteristicDiscovered().Subscribe(obj =>
            {
                mCharacteristics.Add(new CharacteristicBLE(obj, CharateristicReady));
            });

//When a characteristic is added this is run
            if (pInput.CanNotify())
            {
                snv = Input.SetNotificationValue(CharacteristicConfigDescriptorValue.Indicate).Subscribe((temp) =>
                {
                    wnr = pInput.WhenNotificationReceived().Subscribe(obj =>
                    {
                        var buffer = obj.Data;
                        var charEvent = new CharacteristicEvent(buffer);
                        _ValueChanged?.Invoke(obj, charEvent);
                    });
                });
            }

No references are stored to actual services or characteristics (i read your notes on the topic).

SeppyMoose, 2017-06-29:
This issue is fixed with:
```C#
                snv = 
pInput.SetNotificationValue(CharacteristicConfigDescriptorValue.Indicate).Subscribe((temp) =>
                {
                    Debug.WriteLine("Setting up notify. : " + temp.ToString());
                    wnr = pInput.WhenNotificationReceived().Subscribe(obj =>
                    {
                        Debug.WriteLine("Charateristic Notify Event From : " + pInput.Service.Device.Uuid.ToString());

                        var buffer = obj.Data;
                        var charEvent = new CharacteristicEvent(buffer);
                        _ValueChanged?.Invoke(obj, charEvent);
                    });
                });
                **pInput.SetNotificationValue(CharacteristicConfigDescriptorValue.None);**
```

aritchie, 2017-06-29:
//THIS LINE, NOT SURE WHAT IT DOES
                pInput.SetNotificationValue(CharacteristicConfigDescriptorValue.None);

Turns off notifications.  It used to be that you could just unsubscribe from the notification to cancel everything, but I needed to be able to tell when the subscription was ready, hence SetNotificationValue.

You can use the extension method characteristic.RegisterAndNotify with will SetNotificationValue to start and stop.  For the next minor release, I may build an overloaded argument back into WhenNotificationReceived()

SeppyMoose, 2017-06-29:
I actually didn't seem to fix the issue, i still cannot communicate with multiple devices consistently with the above code. Sometimes the first one continues to  communicate, sometimes the last one, very occasionally both continue.

aritchie, 2017-06-29:
You aren't awaiting/subscribing to the pInput.SetNotificationValue(CharacteristicConfigDescriptorValue.None);

Just do: await pInput.SetNotificationValue(CharacteristicConfigDescriptorValue.None);

SeppyMoose, 2017-06-29:
I'm avoiding any async code, I'd prefer a continue with solution or subscribe solution, not sure what to do in that regard.
I think i understand though, can I just do the following:
```C#
                pInput.SetNotificationValue(CharacteristicConfigDescriptorValue.None).Subscribe((temp) =>
                {
                    Debug.WriteLine("Setting up notify. : " + temp.ToString());
                    wnr = pInput.WhenNotificationReceived().Subscribe(obj =>
                    {
                        Debug.WriteLine("Charateristic Notify Event From : " + pInput.Service.Device.Uuid.ToString());

                        var buffer = obj.Data;
                        var charEvent = new CharacteristicEvent(buffer);
                        _ValueChanged?.Invoke(obj, charEvent);
                    });
                });```

aritchie, 2017-06-29:
You missed what I said.  Do the following:

1) Subscribe to notifications pInput.WhenNotificationReceived().Subscribe(...);  You can and should do this before enabling notifications.

2) You have to enable notifications: pInput.SetNotificationValue(CharacteristicConfigDescriptorValue.Indicate);  subscribe or await

3) When you are done listening, disable notifications pInput.SetNotificationValue(CharacteristicConfigDescriptorValue.None)
You are getting the additional broadcasts likely because you are not disabling the notifications.

OR

Use the extension method I suggested: characteristic.RegisterAndNotify().Subscribe(...)

SeppyMoose, 2017-06-30:
I am doing all of the first list:
```C#
            if (pInput.CanNotify())
            {
                snv = pInput.SetNotificationValue(CharacteristicConfigDescriptorValue.Indicate).Subscribe( (temp) =>
                {
                    Debug.WriteLine("Setting up notify. : " + temp.ToString());
                    wnr = pInput.WhenNotificationReceived().Subscribe(obj =>
                    {
                        Debug.WriteLine("Charateristic Notify Event From : " + pInput.Service.Device.Uuid.ToString());
                        var buffer = obj.Data;
                        var charEvent = new CharacteristicEvent(buffer);
                        _ValueChanged?.Invoke(obj, charEvent);
                    });
                });
            }
```

I tried to replace 
```C#
            if (pInput.CanNotify())
            {
                snv = pInput.RegisterAndNotify().Subscribe( (obj) =>
                {
                        Debug.WriteLine("Charateristic Notify Event From : " + pInput.Service.Device.Uuid.ToString());
                        var buffer = obj.Data;
                        var charEvent = new CharacteristicEvent(buffer);
                        _ValueChanged?.Invoke(obj, charEvent);
                });
            }
```

I'm running the above and no notifications are received, I can confirm that the function is run and should have received notifications but doesn't.

aritchie, 2017-06-30:
1) RegisterAndNotify has a default value of Notify, not Indicate.  Does the characteristic support both?  Usually, they don't.

2) With the original                 snv = pInput.SetNotificationValue(CharacteristicConfigDescriptorValue.Indicate).Subscribe( (temp) =>

That temp variable in the subscribe returns if the enabling of the broadcast was successful.  You need to check it.

If you are still having problems, you will need to submit a full test case.

SeppyMoose, 2017-06-30:
Yeah it doesn't support both, just indications.
Both characteristics successfully subscribe and might even indicate a couple times simultaneously. After that it seems like the last added device dominates and is the only one to get indications.

SeppyMoose, 2017-06-30:
Here is a full test case:
https://gitlab.com/Sepps/aritchie-bluetoothle-testcase

To use it, get two ble devices that indicate, select them from the on-screen listview one by one, wait a few seconds between them.

Messages will appear in the debug output that indicates when indications appear.

SeppyMoose, 2017-06-30:
Issue isolated to older versions (5.1). The code functions perfectly in 6+ android.

aritchie, 2017-06-30:
Somehow this issue has changed topics several times.  In any case, on Android, it is possible to overwhelm the device with a flood of notifications.  There is nothing I found to help with this.  I'm not doing anything magical under the hood either. 

Since your initial issue is now more or less solved, I am closing this issue.

closed 2017-06-30 by aritchie