aritchie/bluetoothle #57: Extensions_Adapter.cs

rlingineni, 2017-07-01:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] PCL
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of Library
3.1.1

## Expected Behaviour
ScanWhenAdapterReady(); should continue as a normal scan. Disposing it should stop the scan in progress.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
Scanning is not actually stopped when Disposed, when the `ScanWhenAdapterReady();` is used to begin a scan. Throws error message when my scan pages opens a second time `An existing scan is in progress`

aritchie, 2017-07-01:
I ran some tests with it and it worked just fine.  Please provide a test sample with showing the issue

aritchie, 2017-07-24:
No reply - closing

closed 2017-07-24 by aritchie