aritchie/bluetoothle #60: Issue getting notifications from characteristic

toryjleo, 2017-07-27:
## Version of Library
4.0.0

Hello, I am trying to receive notifications from a characteristic, but I have been unsuccessful. It is my understanding, that if I want to get notifications from a characteristic, and use the data from the notification,  I need to call: 

characteristic.SetNotificationValue(CharacteristicConfigDescriptorValue.Notify);
this.rxChar.WhenNotificationReceived().Subscribe(result =>{result.Data ...});

and this to stop getting notifications:

rxChar.SetNotificationValue(CharacteristicConfigDescriptorValue.None);

But, this doesn't seem to be working, and on the main page of this library, you seem to suggest to call:

characteristic.SubscribeToNotifications(result => {
    	//result.Data to get at response
});

which seems to be outdated, for the method "SubscribeToNotifications" isn't in your "IGattCharacteristic" interface. I am sorry for opening this as an issue, but I am really confused.

aritchie, 2017-07-27:
Please use the full issue template in the future if you want my help please.  

I don't know what platform you are on to suggest what the problem may be.  Have you followed the setup portion for the platform(s) in question?  Does your characteristic support notifications or indications - there is a difference?  

You are correct that the main readme is out of date, but you are making the right calls in any case.

toryjleo, 2017-07-27:
My apologies for not using your template. Thanks for the quick response; I spent the afternoon going over my code. It is probably an issue with something I am doing and I am working in a PCL and the characteristic supports notifications.

aritchie, 2017-07-27:
I still don't know how you want me to help you?  I have no idea what platform you are using - android and/or iOS?  You haven't shown me any code really.  The only thing I k ow right now is that "it doesn't work"

closed 2017-07-30 by aritchie