aritchie/bluetoothle #61: Disconnecting and reconnecting to same device causes double connections.

adamrobert, 2017-07-29:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] PCL
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
5.00, 6.00

## Version of Library
4.0.0

## Expected Behaviour
Disconnecting from a device and scanning to reconnect to it.

## Actual Behavior
When connecting to the same ble device after disconnecting and scanning for it again seems to cause double connections.

This is roughly the code I have been using to test it:

        
        private IDevice device;
        private IDisposable scanner; 

        private List<IGattCharacteristic> characteristcs;
        private List<IDevice> results;

		
        results = new List<IDevice>();
        characteristcs = new List<IGattCharacteristic>();
                     
            button1.Clicked += (s, e) =>
            {
                
                scanner = CrossBleAdapter.Current.Scan().Subscribe(scanResult =>
                {
                    Debug.WriteLine("scanResult.Device.Name: " + scanResult.Device.Name + " UUID: " + scanResult.Device.Uuid);
                    if (!results.Contains(scanResult.Device))
                    {
                        results.Add(scanResult.Device);
                    }
                });
                             
            };

            button2.Clicked += (s, e) =>
            {
                
                scanner.Dispose();
                foreach (var d in results)
                {
                    Debug.WriteLine("Device Name: " + d.Name);
                    if (d.Name == "ble device")
                    {
                        device = d;
                        Debug.WriteLine("Success");
                        device.WhenStatusChanged().Subscribe(connectionState =>
                        {
                            Debug.WriteLine("");
                            Debug.WriteLine("Connection State: " + connectionState);
                            Debug.WriteLine("");

                            if (connectionState == ConnectionStatus.Disconnected)
                            {
                                characteristcs.Clear();
                            }
                        });

                        device.WhenAnyCharacteristicDiscovered().Subscribe(characteristic =>
                        {
                            Debug.WriteLine("Description: " + characteristic.Description + " Characteristic :" + characteristic.Uuid);
                            characteristcs.Add(characteristic);
                        });
                    }                    
                } 
                results.Clear();                              
            };
            button3.Clicked += async (s, e) =>
            {
                await device.Connect(new GattConnectionConfig
                {
                    IsPersistent = false
                });
            };
            //used for testing the ble device
            button4.Clicked += async (s, e) =>
            {
                await characteristcs[characteristcs.Count - 1].Write(new byte[] { 0x01 });
            };

            button5.Clicked += (s, e) =>
            {               
                device.CancelConnection();                           
            };
        }

So on the first connection and disconnection, everything works fine its when I scan again and connect to the same device again that the issue arises. The console seems to show there is a double connection and discovery of characteristics happens more than once. After disconnecting one more time, either the ble device will randomly reconnect or a new scan will not show the device as if it is still connected.

// first scan, connection and disconnection
[0:] scanResult.Device.Name: ble device UUID: 00000000-0000-0000-0000-000000000001
[0:] Device Name: ble device
[0:] Success
[0:] 
[0:] Connection State: Disconnected
[0:] 
[0:] 
[0:] Connection State: Connected
[0:] 
[0:] Description:  Characteristic : 1
[0:] Description:  Characteristic : 2
[0:] Description:  Characteristic : 3
[0:] Description:  Characteristic : 4
[0:] Description:  Characteristic : 5 
[0:] Description:  Characteristic : 6
[0:] 
[0:] Connection State: Disconnecting
[0:] 
[0:] 
[0:] Connection State: Disconnected
[0:] 

// second scan, connection
[0:] scanResult.Device.Name: ble device UUID: 00000000-0000-0000-0000-000000000001
[0:] Device Name: TZ Radial
[0:] Success
[0:] 
[0:] Connection State: Disconnected
[0:] 
[0:] 
[0:] Connection State: Connected
[0:] 
[0:] 
[0:] Connection State: Connected
[0:] 
[0:] Description:  Characteristic : 1
[0:] Description:  Characteristic : 2
[0:] Description:  Characteristic : 3
[0:] Description:  Characteristic : 4
[0:] Description:  Characteristic : 5
[0:] Description:  Characteristic : 6
[0:] Description:  Characteristic : 1
[0:] Description:  Characteristic : 2
[0:] Description:  Characteristic : 3
[0:] Description:  Characteristic : 4
[0:] Description:  Characteristic : 5
[0:] Description:  Characteristic : 6

I hope that all makes sense. 

Kind Regards

Adam

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible
Contact GitHub

aritchie, 2017-07-29:
You can't really double connect to most gatt devices.  Are you getting double notifications or actual issues?  Does the problem get exponentially worse for you (3rd connection shows even more connected events)?
  
One thing that could be happening is that the characteristics you have are pinned.  You should clear them every scan (as all of the devices are cleared).

adamrobert, 2017-07-30:
So it appears that it is double notifications if I only ever use the last added characteristics they appear to work and there is only ever one device in the CrossBleAdapter.Current.GetConnectedDevices();.

Unfortunately, it does seem to cause other issues, the main one being that in the second scan and then disconnection, there is another (automatic) connection event but it does not discover the characteristics again, and no matter how many times I try to disconnect it reconnects, and if I scan again the ble device is not shown. Its almost as if it is maintaining a reference somewhere.

So I did some more testing if I remove all references to my characteristics list nothing changes, but once I remove the WhenAnyCharacteristicDiscovered().Subscribe(characteristic => I don't have the weird (automatic) reconnection bug mentioned above still get double connection notifications though (Which do seem to add up depending on how many times I disconnect, scan and connect).

The other thing I tried was instead of scanning for the ble device again and then copying its new reference to connect too, I just called Connect on the original reference, and I have no issues disconnecting and reconnecting to it any number of times, with discovering and using characteristics as well.

So it seems that when scanning for devices again and using the new reference to connect, something to do with the old reference gives extra notifications?

aritchie, 2017-07-30:
You cannot keep references to any thing after a scan (devices, characteristics, or existing subscriptions) or after a device connect/reconnect which is partly covered in the FAQ of the readme.  You also need to dispose of the subscriptions if you don't want them to be hooked since you are never killing this class (in your example) as the GC won't clean them up for you.  

I ran through with the test app in the repository and it did not experience this issue.

adamrobert, 2017-08-01:
Thanks so much, yep it turned out to be a hook in the subscriptions, once I keep references to them and then dispose of them when disconnecting everything works perfectly. 

Thanks again

closed 2017-08-01 by adamrobert