aritchie/bluetoothle #62: UWP. Unhandled exception in case of mac address with leading 0

susloparovdenis, 2017-07-30:
[image]

aritchie, 2017-08-09:
Thanks for the PR and bug.  Can you please provide a MAC address where this was failing?  I haven't had an issue with this function in the current testing

susloparovdenis, 2017-08-09:
08:66:98:D0:3B:6A
but in the address argument was the string "86698D03B6A"

KevinBaselinesw, 2017-09-01:
I hit this problem as well.   Here is how I fixed it.  Feel free to integrate if you would like

    protected Guid GetDeviceId(string address)
        {
            string EmptyGuid = Guid.Empty.ToString();

            address = EmptyGuid.Substring(0, EmptyGuid.Length - address.Length) + address;

            var mac = address
                .Replace("BluetoothLE#BluetoothLE", String.Empty)
                .Replace(":", String.Empty)
                .Replace("-", String.Empty);

            var deviceGuid = new byte[16];
            var macBytes = Enumerable
                .Range(0, mac.Length)
                .Where(x => x % 2 == 0)
                .Select(x => Convert.ToByte(mac.Substring(x, 2), 16))
                .ToArray();

            macBytes.CopyTo(deviceGuid, 0); // 12 bytes here if off the BluetoothLEDevice
            return new Guid(deviceGuid);
        }

commit 68a92faf (2017-09-09): fix #62 - mac address fix for uwp
closed 2017-09-09 by aritchie