aritchie/bluetoothle #64: Could not resolve type with token 01000099 (from typeref, class/assembly System.Reactive.Subjects.SubjectBase`1, System.Reactive.Linq, Version=3.0.3000.0, Culture=neutral, PublicKeyToken=94bc3704cddfc263)

vaskanas, 2017-08-08:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] PCL
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
5.1.0, 6.0.1

## Version of Library
4.0.0

## Expected Behaviour

## Actual Behavior
Exception: Could not resolve type with token 01000099 (from typeref, class/assembly System.Reactive.Subjects.SubjectBase`1, System.Reactive.Linq, Version=3.0.3000.0, Culture=neutral, PublicKeyToken=94bc3704cddfc263)

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible
Contact GitHub

aritchie, 2017-08-08:
This has nothing to do with this library.  It is a Xamarin issue.  Usually, close solution, delete bin/obj folders from your project.  Restart VS and you should be good.

closed 2017-08-08 by aritchie

vaskanas, 2017-08-09:
Thanks for your immediate response, but unfortunately it does not work. However, I tried to use your  library Acr.Ble.Server and everything works fine. So I doubt that it is a Xamarin issue.

aritchie, 2017-08-09:
It is, but I'm not going to debate this fact with you.  I have discontinued Acr.Ble.Server because it was merged into this library. 

In any case, if you've found a suitable solutions, that's great.  Best of luck

x37v, 2018-02-20:
hmm, I get this exception when I try to connect to a device. deleting bin/obj didn't resolve it for me either, I'm not using the Acr.Ble.Server, I'm just trying to connect to a BLE device using a simplified version of the Sample code provided in this repository.

```
using (var cancelSrc = new CancellationTokenSource(10000))
{
    await dev.Connect().ToTask(cancelSrc.Token); //dev is an IDevice, exception thrown here
}
```

aritchie, 2018-02-20:
TALK TO XAMARIN....!  I will restate this one last time and then I’ll just start removing people from being able to post here at all.

x37v, 2018-02-21:
I can see that this is a touchy issue and I've only commented here because this github issue is the only result I see when I do a web-search about the exception I get.

Clearly the issue doesn't exist in the sample application you've provided @aritchie , I wonder if you can provide some information about what of the dependencies we might need to hold back or use some pre-release version? A work around? For instance, I see that the plugin sample uses a pre-release version of the reactiveui dependency, though everything in the call stack that produces the issue seems to be in line with the sample that I have running, from this repository, without issue.

```
System.Runtime.CompilerServices.AsyncTaskMethodBuilder<System.Action>.Start<Plugin.BluetoothLE.Device.<>c__DisplayClass12_0.<<Connect>b__0>d>(Plugin.BluetoothLE.Device.<>c__DisplayClass12_0 stateMachine) in 
Plugin.BluetoothLE.Device.(System.Reactive.AutoDetachObserver<object> ob) in 
System.Reactive.Linq.QueryLanguage.<object>(System.Reactive.AutoDetachObserver<object> observer, System.Threading.CancellationToken token) in 
System.Reactive.Linq.QueryLanguage.<object>(System.Reactive.AutoDetachObserver<object> observer) in 
System.Reactive.AnonymousObservable<object>.SubscribeCore(System.Reactive.AutoDetachObserver<object> observer) in 
System.Reactive.ObservableBase<object>.ScheduledSubscribe(System.Reactive.Concurrency.CurrentThreadScheduler _, System.Reactive.AutoDetachObserver<object> autoDetachObserver) in 
System.Reactive.Concurrency.ScheduledItem<System.TimeSpan,System.Reactive.AutoDetachObserver<object>>.InvokeCore() in 
System.Reactive.Concurrency.ScheduledItem<System.TimeSpan>.Invoke() in 
System.Reactive.Concurrency.CurrentThreadScheduler.Trampoline.Run(System.Reactive.Concurrency.SchedulerQueue<System.TimeSpan> queue) in 
System.Reactive.Concurrency.CurrentThreadScheduler.Schedule<System.Reactive.AutoDetachObserver<object>>(System.Reactive.AutoDetachObserver<object> state, System.TimeSpan dueTime, System.Func<System.Reactive.Concurrency.IScheduler,System.Reactive.AutoDetachObserver<object>,System.IDisposable> action) in 
System.Reactive.Concurrency.LocalScheduler.Schedule<System.Reactive.AutoDetachObserver<object>>(System.Reactive.AutoDetachObserver<object> state, System.Func<System.Reactive.Concurrency.IScheduler,System.Reactive.AutoDetachObserver<object>,System.IDisposable> action) in 
System.Reactive.ObservableBase<object>.Subscribe(System.Reactive.AnonymousObserver<object> observer) in 
System.Reactive.Threading.Tasks.TaskObservableExtensions.ToTask<object>(System.Reactive.AnonymousObservable<object> observable, System.Threading.CancellationToken cancellationToken, object state) in 
System.Reactive.Threading.Tasks.TaskObservableExtensions.ToTask<object>(System.Reactive.AnonymousObservable<object> observable, System.Threading.CancellationToken cancellationToken) in 
```

Can you at least point us at some bug report for the Xamarin issue that you're referencing?