aritchie/bluetoothle #65: Unable to get device name on bluetooth scan

akashpavate58, 2017-08-09:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS

 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

## Version of Library

## Expected Behaviour
```
            var Adapter = CrossBleAdapter.Current;
            
            if(Adapter.Status == AdapterStatus.PoweredOn)
            {
                Scanner = CrossBleAdapter.Current.Scan().Subscribe(scanResult =>
                {
                    ScanResults.Add(new BLEResult {
                        Name = scanResult.Device.Name,
                        RSSI = scanResult.Rssi,
                        LocalName = scanResult.AdvertisementData.LocalName,
                        UUID = scanResult.Device.Uuid
                    });

                });
            }else
            {
                await DisplayAlert("Found", $"Status: {Adapter.Status}", "Ok");
            }
```

The above code should have returned the device name.

## Actual Behavior
The code gets the RSSI value, but not the Name. How should I search for device names. 

Requirement: 
I have 5 BLE Beacons namely beacon1, beacon2, beacon3, beacon4 & beacon5. I want to pool for these beacons RSSI strength continuously. 

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible
Contact GitHub

aritchie, 2017-08-09:
When it comes to beacons with iOS, the best advice I can give you is to work with them via the beacon APIs, not bluetooth (corebluetooth).  Apple tends to intercept A LOT of the beacon packets for its on purposes.  

As an FYI, RSSI is useless unless you know the TxPower (which is usually a constant) and apple 100% rips that out the beacon packet.

akashpavate58, 2017-08-09:
Ok... Can I at least search for phone around me? Like the names

aritchie, 2017-08-09:
As long as the phone is advertising something that doesn't resemble an ibeacon- yes

closed 2017-08-09 by aritchie