aritchie/bluetoothle #66: Device disconnect on GetKnownService or GetKnownCharacteristic crashes app

mariusbloemhof, 2017-08-14:
Hey Archie, hope you are well,

I have encountered a bug which I cannot seem to get around.

A real world scenario for our device is it disconnecting and reconnecting frequently, especially when on the edge of the signal range.

Therefore it is possible that the device can disconnect while busy obtaining services or characteristics or anywhere in between. We have simulated such a scenario by switching off our device as soon as either GetKnownService or GetKnownCharacteristic is called. Invariably this causes our app to crash, and no matter how we try, this error cannot be trapped, it simply bypasses any exception catching. See snippet below. I have also tried using the OnError action of the subscribers, but the app simply crashes.

```
            try
            {
                ConnectedDevice?.GetKnownService(Guid.Parse("0000ffe0-0000-1000-8000-00805f9b34fb")).Subscribe(server => 
                {
                    Console.WriteLine("GetKnownService");
                    try
                    {
                        server.GetKnownCharacteristics(Guid.Parse("0000ffe1-0000-1000-8000-00805f9b34fb")).Subscribe(characteristic =>
                        {
                            Console.WriteLine("GetKnownCharacteristics");
                            try
                            {
                                DoAuthorize(characteristic);
                            }
                            catch (Exception e)
                            {
                                Console.WriteLine("GetKnownCharacteristics: ", e.Message);
                            }

                        });
                    }
                    catch (Exception e)
                    {
                        Console.WriteLine("GetKnownService: ", e.Message);
                    }

                });

            }
            catch (Exception e)
            {
                Console.WriteLine("DoGetCharacteristics: ", e.Message);
            }
```

[stacktrace.txt](https://github.com/aritchie/bluetoothle/files/1223454/stacktrace.txt)

mariusbloemhof, 2017-08-14:
Stacktrace attached

aritchie, 2017-08-15:
I see the issue.  It is a race condition with iOS that was left out of the "known" characteristic retrieval method.

If you do a .Catch<IGattCharacteristic>(Observable.Empty<IGattCharacteristic>()).Where(x => x != null) that will swallow the error and filter it out

aritchie, 2017-08-15:
4.0.1 should have the fix you need

closed 2017-08-15 by aritchie

mariusbloemhof, 2017-08-15:
Awesome - works like a charm!