aritchie/bluetoothle #68: Getting no device list on bluetooth scan

aloksharma1, 2017-08-22:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] PCL
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
from 4.3 to 6.0 i tried 4 devices

## Version of Library
latest acr ble 2.0

## Expected Behaviour
i am trying to follow given example on xamarin.android to get and connect/pair with avail devices.

here is the code sample i am trying.

using Android.App;
using Android.Widget;
using Android.OS;
using System;
using Plugin.BluetoothLE;
using Plugin.BLE;

namespace BlE_Communications
{
    [Activity(Label = "BlE_Communications", MainLauncher = true, Icon = "@drawable/icon")]
    public class MainActivity : Activity
    {
        protected override void OnCreate(Bundle bundle)
        {
            base.OnCreate(bundle);
            SetContentView(Resource.Layout.Main);
            Button btnSend = FindViewById<Button>(Resource.Id.btnSend);
            btnSend.Click += onbtnSendClick;
            // Set our view from the "main" layout resource
            // SetContentView (Resource.Layout.Main);
        }
        private async void onbtnSendClick(object sender, EventArgs e)
        {
            Toast.MakeText(context: this, text: "scanning", duration: ToastLength.Long).Show();
            //if (CrossBleAdapter.Current.CanOpenSettings())
            //    CrossBleAdapter.Current.OpenSettings();
            if (CrossBleAdapter.Current.CanControlAdapterState())
                CrossBleAdapter.Current.SetAdapterState(true);
            // discover some devices
            TextView tvStatus = FindViewById<TextView>(Resource.Id.textView1);           
           
            /*the below code is giving null devices*/
            //if (CrossBleAdapter.Current.Status == AdapterStatus.PoweredOn)
            //{
            //    CrossBleAdapter.Current.Scan().Subscribe(scanResult =>
            //{
            //    tvStatus.Text += System.Environment.NewLine + scanResult.Device.Name;
            //    tvStatus.Text += System.Environment.NewLine + scanResult.AdvertisementData.IsConnectable;
            //    tvStatus.Text += System.Environment.NewLine + scanResult.Device.Uuid;
            //    tvStatus.Text += System.Environment.NewLine + scanResult.Rssi;
            //});
            //}
           /*tried this one from documentation too but still getting no device*/
            var devices = CrossBleAdapter.Current.GetPairedDevices();
            foreach (var device in devices)
            {
                tvStatus.Text = System.Environment.NewLine + device.Name;
            }

            //tvStatus.Text = "After Scan";
            // Once finding the device/scanresult you want

            //Device.WhenAnyCharacteristicDiscovered().Subscribe(async characteristic =>
            //{
            //    // read, write, or subscribe to notifications here
            //    var result = await characteristic.Read(); // use result.Data to see response
            //    await characteristic.Write(bytes);

            //    characteristic.SubscribeToNotifications(result =>
            //    {
            //        //result.Data to get at response
            //    });
            //});
        }

    }
}

aritchie, 2017-08-22:
You can try a few things 
1) Check your permissions in your app manifest
2) Keep a reference to scan observable var disposable = CrossBleAdapter.Current.Scan().Subscribe(...)
3) Note that the subscribe callback will fire off the main thread.  You need to dispatch back to the main thread if you want to see UI changes.

aloksharma1, 2017-08-22:
hi,

thanks for prompt response, i checked the issues with permission and its set as it should be.
can you elaborate on point 3 on how to dispatch back to main UI, i have done this on winforms but have no idea on xam.android.

aritchie, 2017-08-22:
You can use RunOnUiThread from your activity.  

Please note that this is not a support forum.  Those types of questions should be done externally to this library in the future.

aloksharma1, 2017-08-23:
hi i tried RunOnUiThread & the disposable var, but it was of no avail. i still cannot see any device that have bluetooth on. while the in-built settings page of android shows the devices without issues.

aritchie, 2017-08-23:
Well - it works for me in the projects I'm working with, so please provide an end-to-end project showing the issue.

aritchie, 2017-08-28:
No response - closing until necessary details have been provided

closed 2017-08-28 by aritchie