aritchie/bluetoothle #71: GattCharacteristic: Read - Object reference not set to an instance of an object

mariusbloemhof, 2017-09-06:
Hi Allan,

We currently in testing with our app and custom device. I am happy to report that so far using your component our BLE is rock solid, and never fails to connect/reconnect etc, even with extensive usage and testing.

I have implemented Azure Mobile Center to log any crashes remotely, and the only error that crops up infrequently is the one below. It crashes our app however. This is, unfortunately, the only info I have available so far. This is never reproduced while debugging and only in the field. 

This seems to be at the point of reading after several disconnect/reconnect cycles:

**Object reference not set to an instance of an object**

1 GattCharacteristic+<>c__DisplayClass11_0.<Read>b__1 (System.Object sender, CoreBluetooth.CBCharacteristicEventArgs args) C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Apple.Shared\GattCharacteristic.cs:89
2 (wrapper delegate-invoke) System.EventHandler`1[CoreBluetooth.CBCharacteristicEventArgs]:invoke_void_object_TEventArgs (object,CoreBluetooth.CBCharacteristicEventArgs)
3 CBPeripheral+_CBPeripheralDelegate.UpdatedCharacterteristicValue (CoreBluetooth.CBPeripheral peripheral, CoreBluetooth.CBCharacteristic characteristic, Foundation.NSError error) /Users/builder/data/lanes/4991/4a279c9a/source/xamarin-macios/src/build/ios/native/CoreBluetooth/CBPeripheral.g.cs:503

Cheers
Marius

aritchie, 2017-09-06:
It looks like the characteristic value is coming up null according to the line number.  Easy enough to fix, but the read always produced an empty array before.

mariusbloemhof, 2017-09-07:
ok thanks. I can handle a null array.

We have had 2 crashes since I reported it - exactly the same. I am kinda glad though because when this crash is resolved our app is ready for release.

I can fix it in the source code myself, but I prefer to use the official release version.

aritchie, 2017-09-09:
Fixed in v5 which has been released to nuget

closed 2017-09-09 by aritchie

mariusbloemhof, 2017-09-11:
Dude you are a legend, thanks so much - all is working perfectly

aritchie, 2017-09-12:
Thanks - glad to hear it works.