aritchie/bluetoothle #72: Android: Unhandled exception in callback for onConnectionStateChange

mariusbloemhof, 2017-09-20:
Android only:

[BLEDevice.zip](https://github.com/aritchie/bluetoothle/files/1318763/BLEDevice.zip)
[callstack.txt](https://github.com/aritchie/bluetoothle/files/1318764/callstack.txt)
 
_SDK: API 26, 
minimum AP level 16_

Hi Allan, hope you are well.

I have observed this exception in my debug output (full callstack attached):

**java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGattCallback.onConnectionStateChange(android.bluetooth.BluetoothGatt, int, int)' on a null object reference**

The error is observed very specifically only when **re-acquiring** characteristics and subscribing to them after a reconnection. I have attached a simple test class which presents this error. In my case, this error occurs when calling Device.Connect(line 140) after the reconnection - the first time everything is 100%. I have set up the test unit to disconnected and then reconnect and resubscribe after 3 notifications have been received. 

P.S. I am happy to report that our IOS implementation has been running nonstop, bugfree and stable for a week on end in testing:)

Cheers
Marius

aritchie, 2017-09-24:
I have not observed the condition in my tests and the test case is a too specific for me to do anything with.  

Please make sure to use the template in the future because the OS where this problem was observed would be very helpful in determining what may be going on

mariusbloemhof, 2017-09-26:
Apologies, I have updated my original post with the information.

I will see if I get it on lower versions as well, but I am pretty sure it is exhibiting it as well since our app was displaying the same symptoms on an API level 19 device as well, which is why I created the test code

Will update post with this confirmation asap.

mariusbloemhof, 2017-10-02:
Hi Allan,

I can report that on API 19 device (Samsung SM-J100H), I get the same error:

> 10-02 21:37:31.991 W/BluetoothGatt(21261): Unhandled exception in callback
> 10-02 21:37:31.991 W/BluetoothGatt(21261): java.lang.NullPointerException
> 10-02 21:37:31.991 W/BluetoothGatt(21261): 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:172)
> 10-02 21:37:31.991 W/BluetoothGatt(21261): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
> 10-02 21:37:31.991 W/BluetoothGatt(21261): 	at android.os.Binder.execTransact(Binder.java:404)
> 10-02 21:37:31.991 W/BluetoothGatt(21261): 	at dalvik.system.NativeStart.run(Native Method)
> 10-02 21:37:31.991 D/BluetoothGatt(21261): onClientConnectionState() - status=0 clientIf=8 device=98:7B:F3:76:81:18

I have done some digging and encounter the following [threads ](https://stackoverflow.com/questions/29758890/bluetooth-gatt-callback-not-working-with-new-api-for-lollipop)as well as a google [issue](https://issuetracker.google.com/issues/37057260)

```
    public class GattContext
       .
       .
        public void Close()
        {
            this.gatt?.Disconnect();
            this.gatt?.Close(); > this seems to be the issue
            this.gatt = null;
        }
```
as per one of the suggestions, I added a delay before calling gatt?.Close() - this did not fix the issue. When I comment out this line, I no longer get the exceptions, but rest of the callbacks does not work.

**from the Google issue:**

DESCRIPTION 
When using BLE and from an Android app is first disconnecting and then closing a BluetoothGatt instance a NullPointerException can happen in BluetoothGatt.java. 
It seems like the issue is that the mCallback is set to null before removing/unregistering the callback. This means that the GattService still thinks the callback is available when calling e.g. onConnectionStateChange which uses the mCallback instance.

aritchie, 2017-10-04:
Thanks for looking into this.  It is going to take some time to figure out a proper fix for this.

mariusbloemhof, 2017-10-11:
I am currently trying to update Visual Studio/Xamarin/Android SDK to use Android 8.0 (API 26) to see if this might potentially alleviate this issue...

aritchie, 2017-10-11:
It won't change a thing.  As an FYI, I've started work on a new implementation of the connection logic.

aritchie, 2017-10-15:
Can you please give the latest dev branch a try to see if you are still having the issue?

mariusbloemhof, 2017-10-16:
Hi Allan,

I am afraid the error persists in the dev branch as per below:

> [0:] Hooking for write response - 0000ffe1-0000-1000-8000-00805f9b34fb
> [0:] Incoming Characteristic Write Event - 0000ffe1-0000-1000-8000-00805f9b34fb
> 10-16 19:35:06.847 D/BluetoothGatt(15544): cancelOpen() - device: 98:7B:F3:76:7E:CF
> [0:] Hooking for write response - 0000ffe1-0000-1000-8000-00805f9b34fb
> 10-16 19:35:06.874 D/BluetoothGatt(15544): close()
> 10-16 19:35:06.875 D/BluetoothGatt(15544): unregisterApp() - mClientIf=7
> 10-16 19:35:06.896 D/BluetoothGatt(15544): onClientConnectionState() - status=0 clientIf=7 device=98:7B:F3:76:7E:CF
> 10-16 19:35:06.901 W/BluetoothGatt(15544): Unhandled exception in callback
> 10-16 19:35:06.901 W/BluetoothGatt(15544): java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGattCallback.onConnectionStateChange(android.bluetooth.BluetoothGatt, int, int)' on a null object reference
> 10-16 19:35:06.901 W/BluetoothGatt(15544): 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:227)
> 10-16 19:35:06.901 W/BluetoothGatt(15544): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
> 10-16 19:35:06.901 W/BluetoothGatt(15544): 	at android.os.Binder.execTransact(Binder.java:573)

However, the sequence and timing are slightly different. This time I get it immediately when I CancelConnection(); previously it seemed to have been on Connect()

I have attached the same point of failure on **dev** and **master**

[master.log](https://github.com/aritchie/bluetoothle/files/1388508/master.log)
[dev.log](https://github.com/aritchie/bluetoothle/files/1388509/dev.log)

aritchie, 2017-10-24:
This should be fixed as of beta2.  Please note, with your original code, you aren't trapping any errors.  It is expected that Connect & any read/write operations can error.  Connect on droid is actually more likely to bubble out an error if necessary with this release.

mariusbloemhof, 2017-10-26:
Evaluating now, will let you know....

mariusbloemhof, 2017-10-30:
So far it looks good. The re-connection is a bit lethargic, but it seems to be stable.  I have specifically implemented error trapping in my test project for any operation, and so far have not observed any.

I have observed that when I manually disconnect and then try to reconnect, as opposed to out of range or radio switched off/on, that sometimes it does not reconnect. I am still doing more testing and will let you know what I find

mariusbloemhof, 2017-10-30:
Hi Allan,

So I have found the following: initially on first connect all is well. 

1. (a) I connect, (b) obtain a characteristic, (c) write values to it and (d) subscribe to notifications.
2. I disconnect (in any way: CancellConnection, out of range, Bluetooth off)
3. Initiate reconnection. It initiates the normal sequence as in (1).
4. (a) and (b)  is successful
5. (c) is partially successful. It actually writes the values (BLEDevice:148) to the characteristic, but none of the Actions is triggered in the observable (onNext, onError, onCompleted). I know the value was written successfully because our device disconnects if it does not receive the passcode in:149, and none of the functions in the firmware is enabled.

below is an extract from the code

```
taskWrite = characteristic.Write(new byte[] { 0xAA, 0x27, 0xEB, 0x55 })
	.Subscribe(
	auth =>
	{
		System.Diagnostics.Debug.Write("Authorized");

		SetNotify(characteristic);
	},

	error =>
	{
		System.Diagnostics.Debug.Write($"error: {error.Message}");
	},
	() =>
	{
		System.Diagnostics.Debug.Write("completed: characteristic.Write");
	});
```

I have attached sample test code and output

[debug-output.txt](https://github.com/aritchie/bluetoothle/files/1428222/debug-output.txt)
[BLEDevice.zip](https://github.com/aritchie/bluetoothle/files/1428217/BLEDevice.zip)

aritchie, 2017-10-30:
Thanks @mariusbloemhof - your testing has been of amazing help.  Definitely the best I've experienced as an OSS maintainer. 

As for your issue, I'm glad the main connection issue appears to be solved.  I haven't seen the write get locked before.  That is a new one, but I have a reliable & fast way to test it now.  

From looking at your code, there isn't anything that jumps out as an issue especially since you're trapping the errors.  I'll see what I can find.

mariusbloemhof, 2017-10-30:
It's a pleasure - working in the industry I know what massive pain it is when you get an issue that is reported as "I get an error". Even fellow developers, rarely give enough information to track down the actual problem, when it is not really that hard when you get relevant info.

Plus you are doing me a massive favor, and your component is the best and most usable and stable I have found for Xamarin cross-platform, so I can only help you to help us :)

mariusbloemhof, 2017-10-31:
Can I contact outside of this forum if you don't mind? I have a simple, BLE related question I want to ask you, and I am sure you have loads of expertise around this

nablai, 2017-11-01:
does your problems occur on all devices?
(or maybe mostly on Samsung S8? (as i am))

mariusbloemhof, 2017-11-02:
I get this on a Samsung SM-J100H (Android 4.4 - API 19) as well. I have not tested another manufacturer other than Samsung. I will need to get my hands on more phones.

The original phone was an S8 though yes

aritchie, 2017-11-03:
@mariusbloemhof you can contact me on the xamarin slack (xamarinchat.slack.com - username aritchie)

aritchie, 2017-11-05:
master now contains a new connection mechanism ported from rxandroidble project that is supposed to take of all these junk implementations on other phones.  

I haven't had any luck reproducing the write locking you're experiencing.  Could you try modifying the new test projects to see if you experience the issue there?

mariusbloemhof, 2017-11-07:
Hi Allan, so far all looks good on master. I modified your test project to include my authentication, and it worked, even on reconnection events. I then transferred some of the code over to my test project and managed to get it working, Once I have it 100% working I will post the differences and let you know. I will then also transfer the changes to your main app and report back

Cheers!

SeeD-Seifer, 2017-11-07:
We are facing an issue described by @mariusbloemhof in https://github.com/aritchie/bluetoothle/issues/72#issuecomment-340532461.

Are there any estimations when it's possible to try a fixed version from NuGet (maybe preview)?

aritchie, 2017-11-07:
@mariusbloemhof you likely only need to run with the unit tests as they will draw out the worst of the problem.

@SeeD-Seifer it breaks my mojo when people ask for estimates on OSS.  If you could run master and report back feedback, that would be helpful.  The more positive feedback I can get, the better chance of a release.

SeeD-Seifer, 2017-11-07:
@aritchie sure thing! Just prefer to ask if it makes sense to wait another night in case preview release is planned anyway :)

Will try during the week and let you know if master fixes the issue for us.

mariusbloemhof, 2017-11-14:
@aritchie 

I have extensively tested using the latest code. I tried to modify the unit tests, but my custom device makes it slightly more tricky, and eventually I end up with my test code:) (see my test class)

I keep on encountering this error (see screenshot)

**System.Threading.SemaphoreFullException: Adding the specified count to the semaphore would cause it to exceed its maximum count.**

I have attached the full output, callstack and the test class that I use to test. My testing sequence is much the same before:

1. (a) I connect, (b) obtain a characteristic, (c) write values to it and (d) subscribe to notifications.
2. I disconnect (in any way: CancellConnection, out of range, Bluetooth off)
3. Initiate reconnection. It initiates the normal sequence as in (1).
4. (a) (b) and (c) is successful
5. (d) seems to be successful but the error is raised, even though the RegisterAndNotify seems to be successful, it then fails on what seems to be the write with the attached error.

I have noticed that you seem to use a lot of task/await, whereas I specifically use the observables.

[debug.zip](https://github.com/aritchie/bluetoothle/files/1472005/debug.zip)

[image]

aritchie, 2017-11-14:
The exception is easy to fix.  I can't delay the writes on my test device so I'm not seeing the gap between passing the gate and still cancelling the task.  

My use of async is just to cause a sequential flow in droid.  The whole android bluetooth stack requires a serial/sequential pattern for ALL actions coming in/out thus the need for locks.  The library is taking a protocol that should be asynchronous, dealing with it serial, while exposing it as asynchronous.  It's complicated. 

Once again, your testing helps a ton... thanks.  Patch should be incoming in a few hours.

mariusbloemhof, 2017-11-20:
Hi Allan, I am encountering a write lock after the initial write to the characteristic. I can 

1. obtain characteristics, 
2. write values to it successfully
3. control in (2) return, but any action after this write, locks execution.

No error is raised. It is the same test unit as before

[output.txt](https://github.com/aritchie/bluetoothle/files/1488602/output.txt)

goiditdev, 2017-11-21:
I can confirm the issues reported above with unstable behavior of writes (and sometimes reads) with the latest commit, and when I roll back the commit I am back at the problem with the race condition on the semaphore. I temporarily solved this by simply trapping exceptions around the semaphore release in DeviceContext.cs and this solves this problem while also retaining the reads and writes. But with that fix I have observed that it seems possible for multiple characteristics accesses against different BLE device instances to "interfere" with each other: They run at the same time (seemingly) and neither of them completes. I imagine that is related to the semaphore gate.

aritchie, 2017-11-21:
Guys - Android requires things to be done in sequence/serial format.  The library is attempting to solve that for you and failing.  This also helps produce the GATT 133 issue that you likely have experienced.  I believe the key issue with this topic has been solved.  The read/write locks will be removed in the next version as fixing the overall issue is difficult.  This means your code will need to be responsible making sure things fire and complete in a serial fashion. 

@goiditdev please contribute fixes instead of just talking about what you think the issue is.  It isn't helpful simply adding confirmation.

closed 2017-11-21 by aritchie

goiditdev, 2017-11-21:
@aritchie My apologies. I didn't commit anything because my code change isn't a fix. It just moved the problem somewhere else.