aritchie/bluetoothle #74: Can't open solution once NuGet package installed

esseivan, 2017-09-25:
I installed the NuGet package Plugin.BluetoothLE 5.0.0 on the project.
Once I closed and reoppened VS2017, the solution takes a very long time (5-10 minutes) to open or it even doesn't.

Sometimes more lucky, it opens without any problem but quite randomly.

I used a blank new app from Xamarin and just imported the Plugin.BluetoothLE 5.0.0 package without any other modification.

Any idea to solve it ?

aritchie, 2017-09-25:
Unfortunately, this sits outside of anything to do with this library.  The nuget package isn't doing anything magically.  However, you should note that this library is based on .NET standard (1.x), not PCL.  The default Xamarin Forms template is still based on the old PCL model.  That may be having an impact on what you're seeing.

Please use the issue template in the future as it saves me having to ask the usual questions.

esseivan, 2017-09-25:
And another thing, just after installing the package, it shows an error saying that System.Runtime.InteropServices.WindowsRuntime is imported two times but when I delete the one in references, more errors shows up

aritchie, 2017-09-25:
Please read my previous message before posting further.  It is definitely related to .NET standard.

esseivan, 2017-09-25:
What should I change to get this working ? Shouldn't .NET standard work with PCL ?

aritchie, 2017-09-25:
PCL's will work in .NET standard.  The other way around is hit or miss depending on the profile matchups.  I was suggest taking a look at a few articles on xamarin forms & .net standard to see what's involved.

esseivan, 2017-09-25:
And should I download the source code here to create a project using your library or just importing the nuget package is enough

aritchie, 2017-10-04:
nuget is enough

closed 2017-10-04 by aritchie