aritchie/bluetoothle #77: Connection trouble with Android

MynaBay, 2017-10-07:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

I'll attach:
-File with snippets of code that I run (same on Android and iOS). 
-Working output log
-Failing output log

## Version of OS(s) listed above with issue
API 22/6.0.1 works.  
API 21/5.1.1 Fails. 
API 25/7.0.X Fails

iOS is fine.

## Version of Library
5.0.0. Nuget package in Visual Studio

## Expected Behaviour
To connect reliably. My Android device on API 21 works, and iOS works. All others (that I've tested) fail. 

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

To work the same on all Androids and iOS.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

[Code_Snippets.txt](https://github.com/aritchie/bluetoothle/files/1366294/Code_Snippets.txt)
[Failing Log.txt](https://github.com/aritchie/bluetoothle/files/1366295/Failing.Log.txt)
[Working Log.txt](https://github.com/aritchie/bluetoothle/files/1366296/Working.Log.txt)

aritchie, 2017-10-07:
Please provide exceptions or more of the samples.  The logs aren't telling me much and the code snippet doesn't appear to have anything out of the ordinary.

MynaBay, 2017-10-07:
There aren't any exceptions. It's just spewing Connecting & Disconnected. Any additional debugging tips? I've been an iOS programmer for years, fairly new to Android. About the only information so far is:

[BluetoothGatt] onClientConnectionState() - status=133 clientIf=9 device=F0:03:8C:93:BF:14

Not much I realize...

MynaBay, 2017-10-07:
I'm going to switch from the nuget package to the project. Hopefully it'll give me a little better debugging info. I'll keep you posted.

MynaBay, 2017-10-09:
I tried running the project's sample app and am getting the same behavior. Seems to be bouncing between SetupAutoReconnect's loop (in WhenStatusChanged handler) and GattCallbacks.OnConnectionStateChange (). The latter is getting a 133 from lower layers. 

The gatt variable in GattCallbacks.OnConnectionStateChange () appears to have hit an exception on it's ConnectedDevices field.

I'll attach a screenshot.

MynaBay, 2017-10-09:
[image]

MynaBay, 2017-10-09:
Nevermind. That exception is a red herring as my working device shows the same. But, it gets a "Connected" value that never comes on the failing. Still digging. Anything I should look for in particular?

MynaBay, 2017-10-09:
Using the Android Device Monitor app and running the BluetoothLE sample app I'm seeing this in the log for devices that fail. Does this help any?

10-09 16:25:43.263: E/bt-att(1576): Unsupported transport for background connection
10-09 16:25:43.263: E/bt-btif(1576): bta_gattc_init_bk_conn failed

I'll attach working and failing logs from AMD shortly.

MynaBay, 2017-10-09:
[Failed to Connect.txt](https://github.com/aritchie/bluetoothle/files/1370101/Failed.to.Connect.txt)

Log from Android Device Monitor for failing device.

MynaBay, 2017-10-09:
[Worked.txt](https://github.com/aritchie/bluetoothle/files/1370103/Worked.txt)

Log from Android Device Monitor for working device.

MynaBay, 2017-10-09:
Just to be clear. This is running the BluetoothLE sample app from GitHub, not our code from the start of this bug.

aritchie, 2017-10-10:
Thank you for the logs.  I'll take a look.

The GATT133 errors are painful to catch since they are usually android device/OS flavour specific due to some weird race condition in the crapshow that is android bleutooth.  The unfortunate fix for these is usually putting a task.delay after the connection before firing all of the events for other things to take off.  Trying to reproduce this will be difficult since it doesn't happen between the 4 droid devices I have, so I wouldn't be waiting for a quick fix here.

aritchie, 2017-10-10:
The make/models and their OS versions of the failing devices would be helpful.  I generally tend to see the bad behaviours on Sony and LG phones.

MynaBay, 2017-10-10:
Failing devices:
    Galaxy S5, model SM-G900T, Android 6.0.1.
    Vizio, Model XR6M10, Android 5.1.1

More to come when I talk with QA.

MynaBay, 2017-10-10:
One thing I discovered is that the using the "PerformActionsOnMainThread" doesn't always work as I think it's expected to. I tracked this down to a couple places where the code is testing "MainThreadSuggested" instead. So, even with PerformActionsOnMainThread set to true, some calls can happen off main. Fixing this locally didn't resolve the problem for me.

aritchie, 2017-10-11:
Please stick to positive findings or requested information to prevent flooding the issue.  The perform main thread is only necessary in some places. 

Thanks for providing the devices.  I'll start with those for now.

aritchie, 2017-10-15:
Can you please give the latest dev branch fixes a try.  I wasn't able to reproduce the original issue on my device collection, but I have no doubt that the problem is there.  I do believe the main issue here is to do with auto-reconnect.

MynaBay, 2017-10-15:
The dev branch appears to have fixed the aforementioned PerformOnMain and the closed out bug on runaway auto-reconnect. It does not fix the reason this bug was opened as I mentioned when I fixed the two issues locally. 

The device does enter the Connected state, however, it's not really connected. I have insight to the BLE accessory side as well. On iOS and working Android devices the BLE accessory reports connected on its console. On failing Android devices with either Dev branch or my local fixes (likely different but same result), Android reports connected while the accessory is still waiting for a connection. The request to GetKnownServices() never finds anything. 

Interestingly, the scan to find the BLE accessory provided a custom service UUID and it's finding the correct device. So, the information is there as it's finding the right accessory by service UUID. Upon querying for that service or other services known to be present there is no response on the failing devices. iOS and working Android behave as expected.

I've found reports of Android devices not finding custom service UUIDs, but have yet to discover a solution. In my case, the BLE accessory is not in agreement about the connection and likely/hopefully the cause of my problem. 

Tools like nRF Connect (both iOS and Android) have no issue connecting, discovering services/characteristics and communicating.

Here are some working and failing log snippets. Not worth much but shows what I reported above. It's something about the connection that isn't right.  Still digging.  I've annotated with "<-------" below.

**FAILING EXAMPLE    Samsung SM-G900T (API 23)  6.0.1**

[Android Side]
	[BluetoothAdapter] STATE_ON
	[BluetoothAdapter] STATE_ON
	[BluetoothLeScanner] onClientRegistered() - status=0 clientIf=6
	[ScanRecord] parseFromBytes
	[ScanRecord] first manudata for manu ID
	Found device: Telo_001861394BB3
	TeloAirBLE.Start()
	[BluetoothManager] getConnectionState()
	[BluetoothManager] getConnectedDevices
	TeloAirBLE.ConnectionStatus: "Telo_001861394BB3" is Disconnected
	[BluetoothGatt] connect() - device: F0:03:8C:93:BF:14, auto: false
	[BluetoothGatt] registerApp()
	[BluetoothGatt] registerApp() - UUID=1e343c0c-7fc9-4819-bb55-8e6746263fa9
	[BluetoothGatt] onClientRegistered() - status=0 clientIf=7
	[ScanRecord] parseFromBytes
	[ScanRecord] first manudata for manu ID
	[BluetoothGatt] onClientConnectionState() - status=0 clientIf=7 device=F0:03:8C:93:BF:14
	Thread started:  #7
	[BluetoothManager] getConnectionState()
	[BluetoothManager] getConnectedDevices
	TeloAirBLE.ConnectionStatus: "Telo_001861394BB3" is Connected        <----- Shows connected
	TeloAirBLE.ConnectionStatus: attempting get service       <-------- GetKnownService call
	[BluetoothGatt] discoverServices() - device: F0:03:8C:93:BF:14
	[BluetoothGatt] onSearchComplete() = Device=F0:03:8C:93:BF:14 Status=0

[BLE Accessory side]
	bleclient: state changed fsm_idle -> fsm_connect_wait				 <------ Still waiting for connection

**WORKING EXAMPLE    Samsung SM-T700 (API 23)  6.0.1**

[Android Side]
	[BluetoothAdapter] STATE_ON
	[BluetoothLeScanner] onClientRegistered() - status=0 clientIf=7
	[ViewRootImpl] ViewPostImeInputStage processPointer 0
	[ViewRootImpl] ViewPostImeInputStage processPointer 1
	[ScanRecord] parseFromBytes
	[ScanRecord] first manudata for manu ID
	[ViewRootImpl] #3 mView = null
	Found device: Telo_001861394BB3
	TeloAirBLE.Start()
	[BluetoothManager] getConnectionState()
	[BluetoothManager] getConnectedDevices
	TeloAirBLE.ConnectionStatus: "Telo_001861394BB3" is Disconnected
	[ViewRootImpl] sendUserActionEvent() mView == null
	[BluetoothGatt] connect() - device: F0:03:8C:93:BF:14, auto: false
	[BluetoothAdapter] isSecureModeEnabled
	[BluetoothGatt] registerApp()
	[BluetoothGatt] registerApp() - UUID=5ef9bded-fc1d-4a64-8ad2-18e5bd7a9a7f
	[BluetoothGatt] onClientRegistered() - status=0 clientIf=9
	[BluetoothGatt] onClientConnectionState() - status=0 clientIf=9 device=F0:03:8C:93:BF:14
	Thread started:  #9
	[BluetoothManager] getConnectionState()
	[BluetoothManager] getConnectedDevices
	TeloAirBLE.ConnectionStatus: "Telo_001861394BB3" is Connected         <------- Shows connected
	TeloAirBLE.ConnectionStatus: attempting get service         <-------- GetKnownService call
	[BluetoothGatt] discoverServices() - device: F0:03:8C:93:BF:14
	[BluetoothGatt] onClientConnParamsChanged() - Device=F0:03:8C:93:BF:14 interval=54 status=0
	[BluetoothGatt] onClientConnParamsChanged() - Device=F0:03:8C:93:BF:14 interval=6 status=0
	[BluetoothGatt] onClientConnParamsChanged() - Device=F0:03:8C:93:BF:14 interval=54 status=0
	[BluetoothGatt] onSearchComplete() = Device=F0:03:8C:93:BF:14 Status=0
	Thread started:  #10
	TeloAirBLE.ServiceFound: c2921afe-f369-49fa-91cf-2760fd66e03d           <--------- Found service
	TeloAirBLE.CharacteristicFound - c2921afe-f369-49fa-91cf-2760fd66e03d   <--------- Found characteristic
	[BluetoothGatt] setCharacteristicNotification() - uuid: c2921afe-f369-49fa-91cf-2760fd66e033 enable: true
	Thread started:  #11
	Queue command: get_myx                                                  <---------  Communicating

[BLE Accessory side]
	bleclient: LE device connected
	bleclient: state changed fsm_connect_wait -> fsm_connected             <------ Shows connection
	bleclient: notifications enabled

alberk8, 2017-10-23:
Hi,
I am getting system crash when connecting using GetKnownDevice.  This occur when the BLE device is not nearby when the code first execute.  

I am using the Plugin.BluetoothLE version 5.1.0-beta1 on Android 5.0 on Samsung Note 3.
[BluetoothLE.cs.txt](https://github.com/aritchie/bluetoothle/files/1405863/BluetoothLE.cs.txt)
[BluetoothLE_ErrorLog.txt.txt](https://github.com/aritchie/bluetoothle/files/1405864/BluetoothLE_ErrorLog.txt.txt)

aritchie, 2017-10-23:
@Alberk8 you need to trap errors.  Connect and characteristic calls will throw exceptions.  I’m currently not using native auto connect under the hood at the moment, so you will need to ensure device availability or write a connect retry

aritchie, 2017-10-24:
Please give 5.1-beta2 on nuget a try.

@alberk8 - the new version has an autoconnect flag that is defaulted to true that should help with your use-case

alberk8, 2017-10-24:
Hi,

I have added the Autoconnect flag but it is spewing a lot of connecting in quick succession until the device comes in range.   The device was first connected then I move it out of range and then back to range.

[BluetoothLE_Error1.txt](https://github.com/aritchie/bluetoothle/files/1409444/BluetoothLE_Error1.txt)

aritchie, 2017-11-05:
I have ported some connection help from the rxandroidble project into master.  Please try it out and let me know.  I'll leave this open for another few days, but after this, if you still experiencing issues - you will need to participate in the debugging and fixing (thereby contributing back to this project instead of just fixing locally) if I'm to go any further with this.

closed 2017-11-14 by aritchie