aritchie/bluetoothle #78: Auto reconnect can swamp a device.

MynaBay, 2017-10-10:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

The issue is in AbstractDevice and should happen in all platforms. I've only reproduce on Android as that's the only platform I'm having connections issues with.

In the function SetupAutoReconnect(), in file AbstractDevice.cs it adds a subscription to WhenStatusChanged().

In this subscription handler is this loop: 

               while (!this.cancelReconnect && this.Status != ConnectionStatus.Connected)
                    {
                        try
                        {
                            await Task.Delay(300);
                            if (!this.cancelReconnect)
                                await this.Connect(config);
                        }
                        catch (Exception ex)
                        {
                            Log.Out("Failed to reconnect to device - " + ex);
                        }
                    }

What's happening is when a device becomes "Disconnected", a thread enters this loop and calls Connect(config). If that connection attempt fails, the status will drop back to "Disconnected". Another thread will enter the loop since the status changed, but the first thread hasn't exited.  

The first failure will have one "loop" attempting to reconnect. Second failure has two, then four, eight, etc..

Very quickly it's swamping a device with connection attempts.

 

## Version of OS(s) listed above with issue

## Version of Library

## Expected Behaviour

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-10-11:
This is likely related to your original issue.  Please don't open any additional defects regarding your connection issue on android

closed 2017-10-11 by aritchie

MynaBay, 2017-10-11:
This isn't really related. The original issue is that it's going to "Disconnected" when "Connected" is expected. I discovered this issue when I turned off IsPersistent and wrote my own reconnect handler. I noticed that mine behaved very differently from the built-in one. A little investigation revealed the root of this issue. 

Neither the built-in reconnect, my reconnect, nor a fixed built-in reconnect handler resulted in a "connected" stated. Still chasing that issue.

If you want to clump them together that's fine. I only wrote separately so this could be fixed and closed. Doesn't help issue 77 though.

agouriou, 2018-02-23:
Hi @MynaBay , I'm experiencing the same issue with the "disconnected" state. 

Did you manage to find a workaround?

MynaBay, 2018-03-05:
@agouriou - Sorry for delay - I've been away. We did find a solution. I believe all the issues have been addressed by aritchie. I'll verify against our codebase.